libimobiledevice/libimobiledevice #679: Method to Completely wipe data without any firmware

yashsharma45, 2018-06-30:
Hello

Is there any method or command to completely wipe data and reboot device like factory reset. I have seen some softwares doing that but i am unable to figure out what command it is or what code i have to add into binary file to do that.

Please reply,

Thanks in Advance

abdulzeedo, 2018-10-02:
There is no way to do that. All the solutions out there do make use of the different firmwares to be able to do that. They have all the firmwares available on their server - once you decide to factory reset the device, the app will get the current firmware version, download it from the server, and install it. 

As simple as that...

yashsharma45, 2018-10-03:
I think there should be another method using Apple Configurator.

What if we want to use same firmware which is currently used by device and we just have top reinstall it without any download from any server.

I think this is possible but i don't know how to do that in LibiMobileDevice.

Waiting for your response.

abdulzeedo, 2018-10-03:
Have your tried it using apple configurator? I am not 100% sure, but even apple configurator might be using this same strategy...