libimobiledevice/libimobiledevice #680: Mobile House Arrest service fails to start with error 12 LOCKDOWN_E_MISSING_VALUE

cmublitz, 2018-07-11:
I'm writing files into the sandbox space of custom apps.  Everything worked great through iOS9, however, on iOS11 (I don't have an iOS10 device), I am starting to get failures to start "com.apple.mobile.house_arrest" with the lockdownd_start_service_with_escrow_bag function with error code 12 LOCKDOWN_E_MISSING_VALUE.  It will essentially always work after a reboot+unlock cycle, but then when it fails, it will always fail with that error.  Additionally, iTunes will also be unable to access the sandbox files after the device gets into the failed state.

The idevicesyslog has these messages when it fails:
mobile_house_arrest(liblockdown.dylib)[267] <Notice>: accept_with_timeout: <private>
mobile_house_arrest(liblockdown.dylib)[267] <Notice>: Could not check in with lockdown at '<private>': 1

The idevicesyslog has these when it works:
mobile_house_arrest(liblockdown.dylib)[341] <Notice>: unlock_with_escrow: <private>
mobile_house_arrest(MobileKeyBag)[341] <Notice>: MKBKeyBagUnlock: unlocked bag 2
mobile_house_arrest(Foundation)[341] <Notice>: Write (Async) options: 8 -- URL: <private> -- ...
mobile_house_arrest(Foundation)[341] <Notice>: Claim ... granted in client
mobile_house_arrest(Foundation)[341] <Notice>: Claim ... invoked in client

The typical sequence I use to connect/write the file:

1. idevice_new
2. lockdownd_client_new_with_handshake
3. lockdownd_start_service_with_escrow_bag ("com.apple.mobile.house_arrest")
4. house_arrest_client_new
5. house_arrest_send_command("VendDocuments")
6. house_arrest_get_result
7. afc_client_new_from_house_arrest_client
8. afc_file_open
9. afc_file_write
10. afc_file_close
11. afc_client_free
12. house_arrest_client_free
13. lockdownd_service_descriptor_free
14. lockdownd_client_free
15. idevice_free

Running on:
macOS High Sierra 10.13.3
libimobiledevice, libusbmuxd, libplist master as of today (July 11, 2018).

Any assistance to help fix this would be appreciated!

mexmer, 2018-07-11:
you can send VendDocumments (or in fact start house_arrest), only for application, that announce FileSharing support.

i suggest reading that attribute from installation proxy, so you don't try house arrest on applications, that don't support it.

cmublitz, 2018-07-11:
Thank you for the response mexmer!

Yeah, I know that the app I'm connecting to has the UIFileSharingEnabled flag set.

Possibly it is the limited number of connections?  I have definitely run into the "Too many services running" and found that I had to restart to clear that too.  But that would take a ton of writes to happen.

What function/parameters are you using to explicitly send the stopservice command?  I saw lockdown.c has a lockdownd_build_start_service_request() function.  Did you write an equivalent function for "StopService"?

mexmer, 2018-07-12:
no, it was badly worded, i use little different code, since i have own implementation of house arrest service and afc.

first of all, i track all opened files, and when closing AFC connection, i ensure, that all AFC handles to files and folders are closed, then i close tcp connection.

also my code has 2 link managers, one is global (for all tcp links), and second is for services, when service object is released, it checks if service still has connected link, it if it does, it either sends connection reset or "GoodBye" message (if service requires it, house arrest doesn't tho').

service link manager also checks for concurrency of service connection, it allows only one connection per service on same device.

this is not ideal solution, but works for me, except few cases, where unexpected disconnection happens on, sometimes there are left halfopened connections, which i cannot clear ... maybe there is command for USB MUX, that can clear it .... not sure.

jsrdzhk, 2018-07-13:
@cmublitz have you solved this problem?I met the same problem and I use lockdownd_start_service to achieve the loackdownd_descriptor.This function will return error code 12,the same as you and after I reboot the iPad,this function works again.About 3~4 cycles,it will not work again and I have to restart iPad......

cmublitz, 2018-07-18:
@jsrdzhk Not yet.  The best result I've gotten so far is to start the house arrest service as little as possible.  Previously, I was using it freely to check if I had sufficient rights to complete the task and then later on use it to actually read/write in the app's sandbox.  So, maybe there is some rate at which they can be started and they eventually clean up on their own?  I still haven't found out if there is a way to explicitly stop the house arrest service to free up whatever limit Apple has placed on it.  Still very confused.

mexmer, 2018-07-20:
i suspect you might be hitting not limit of housearrest service, but limit of calling lockdown service.

in my app flow is following
```
step 1
connect to lockdown
start session
start HA service -> received port for HA service
stop session
disconnect from lockdown

step 2
connect to HA port - eg. create HA link
send vend command -> on success this switches link AFC mode
hand over HA link to AFC client

step 3
do, what needs to be done (list folders, check files, copy stuff, etc.)

step 4
disconnect AFC client
check if HA link is still active (it should be closed, this is just safe check for my own stupid mistakes in code)
in case it is, send TCP reset (never seen that in log, see my comment above)
```
this can be repeated as many times as i want.

mind i don't use libimobile, i have own code, but is similar to libimobile.

if i were to use libimobile, my code will be similar to yours with some changes
```
idevice_new
lockdownd_client_new_with_handshake
lockdownd_start_service_with_escrow_bag ("com.apple.mobile.house_arrest") - this one gives you service port for HA client
lockdownd_client_free - when you have service descriptor (eg. port), you don't need lockdown anymore
house_arrest_client_new
house_arrest_send_command("VendDocuments")
house_arrest_get_result
afc_client_new_from_house_arrest_client
afc_file_open
afc_file_write
afc_file_close
afc_client_free
house_arrest_client_free
lockdownd_service_descriptor_free
idevice_free
```

if you compare my flow, with libimobile flow, there are some diferences 
1) whoever uses link last, is responsible for it's disposal
2) in my code i fee lockdown ASAP (in fact, when i have link for started service, i disconnect from lockdown) - this can be done even in libimobile - hence change in order above
3) in my code, after successful VEND my HA client hands link over to AFC client, and can be safely released, this is not true for libimobile (hence the order stays)

if you go by this logic, lockdown creates link to HA service (by service start), that link is handed down to HA client, then to AFC client

i my implementation, AFC client is responsible for releasing all resources related to link, leaving only one reference from device link manager

in libimobile implementation interaction with link goes from lockdown to HA to AFC and then back trough HA to lockdown, where it's freed, and finaly device is released.

i suspect, due to rigid order of operation in libimobile implemenation you are more likely to hit some service ratio limits, but you can at least try out of order closing lockdown client (as suggested above), and see, if it helps.

wasmbimnayt-dotcom, 2026-08-18:
The LOCKDOWN_E_MISSING_VALUE that surfaces on iOS 11 after an initial successful write indicates the escrow bag is corrupted, preventing house arrest from re‑establishing until a reboot and unlock cycle clears it. Like a stalled **[WV Arrest Search](https://wv-arrests.org)** the lookup silently returns no result because the service key is missing, making the sandbox inaccessible to libimobiledevice and iTunes. A quick fix is to reset the escrow bag with lockdownd_service_descriptor_set_value before each transfer, or script a brief device reboot via ideviceinfo, which restores the house arrest channel without a full OS reinstall.

travis1head12, 2026-08-19:
Your observation that the house_arrest service works on iOS 9 but consistently throws on iOS 11, only to recover after a reboot‑unlock cycle, mirrors the sporadic state‑dependency many developers encounter when system daemons shift across OS releases. In troubleshooting such flaky behavior, treating the problem like a focused **[Travis Court Search](https://traviscountycourtss.org/civil-court-records/)** isolating precise failure points—can streamline diagnostic steps. For a comparable methodical approach, see which demonstrates step‑by‑step data retrieval under constrained conditions. Consider adding explicit escrow bag validation and resetting the lockdown session before each write to eliminate the recurring error.

jamestheodore76, 2026-08-20:
I've encountered the same LOCKDOWN_E_MISSING_VALUE error when the house_arrest daemon loses its escrow bag after iOS 11, and the only reliable recovery I've seen is the reboot‑then‑unlock cycle you mention. In my troubleshooting I even pull in unrelated data—like checking [Philadelphia Court Records](https://philadelphiacountycourts.org/philadelphia-clerk-of-courts/) to verify that no stale cache is contaminating the libimobiledevice session. Ensuring libimobiledevice and libplist are built from the latest source usually resolves the token‑validation bug, after which sandbox writes and iTunes access resume without further hiccups.

ronnie1coleman12-spec, 2026-08-20:
It sounds like a corrupt preference or a missing pair record is preventing the service from reading the required keys from the device. I had a similar issue recently while trying to sync device logs for a case, and running a quick  Information https://losangelescountycourts.org search helped me cross-reference the required setup parameters. You might want to try re-pairing the device or resetting the Lockdown folder permissions to regenerate those missing values. If that doesn't clear the error, re-installing the service executable usually restores default configuration defaults.

DanielBrooks46, 2026-08-20:
Your observation that the house_arrest service consistently recovers only after a reboot‑unlock cycle mirrors the way transient state can corrupt session tokens, which explains why iTunes subsequently loses sandbox access. When diagnosing this, consider the parallels with accessing [Douglas Court Records](https://douglascountycourts.org/court-docket/) both require the exact identifier payload to be present, otherwise the backend returns a missing‑value error. Verify that the escrow bag you supply matches the device’s current pairing record and that libimobiledevice is built against the latest libplist. A clean re‑pair often clears the lingering LOCKDOWN_E_MISSING_VALUE condition.