libimobiledevice/libimobiledevice #689: can't run idevicediagnostics ioreg

ankitrwt, 2018-08-02:
unable to idevicediagnostics ioreg IOPower/IOPMPowerSource on iPhone6,7,8 its always returns 

<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>IORegistry</key>
	<dict>
	</dict>
</dict>
</plist>

But its working fine with iPhone4,4s,5

ankitrwt, 2018-08-02:
`<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>IORegistry</key>
	<dict>
	</dict>
</dict>
</plist>
`

ankitrwt, 2018-08-02:
> <?xml version="1.0" encoding="UTF-8"?>

> <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" 

> "http://www.apple.com/DTDs/PropertyList-1.0.dtd">

> <plist version="1.0">

> <dict>

	

> <key>IORegistry</key>

> 	<dict>

	

> </dict>

> </dict>

> </plist>

ankitrwt, 2018-08-02:
`<?xml version="1.0" encoding="UTF-8"?>`
`<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" ``"http://www.apple.com/DTDs/PropertyList-1.0.dtd">`
`<plist version="1.0">`
`<dict>`
`	<key>IORegistry</key>
``	<dict>`
`	</dict>`
`</dict>`
`</plist>`

mexmer, 2018-08-02:
apple changed accessibility of this data in later versions of iOS, that's why you can access it on old phones, they have iOS 10 or older.

Blefish, 2018-08-02:
https://github.com/libimobiledevice/libimobiledevice/commit/f37b592788bffcb9a70d211aef8b1f84751114ca

mexmer, 2018-08-02:
@Blefish that's what i meant, since iOS 11 you cannot query plain IORegistry for full list of keys with values, but you actually need to know value name and specify it, when querying ioregistry plane.

mexmer, 2018-08-02:
if anybody knew, list have been already implemented. even apple is quite skimpy, when it comes to information regarding internals.

ankitrwt, 2018-08-03:
@mexmer  i want to know if there is any Network KEY available,,

closed 2018-08-29 by ankitrwt