libimobiledevice/libimobiledevice #690: How to bypass language screen and region screen at activation process (skip assistant set up) ?

truonggiang0710, 2018-08-07:
Hi @nikias , apple configurator 2 can skip language and region over prepare device.
How can we do that ?

truonggiang0710, 2018-09-20:
@jduranc , you can use apple configurator on MAC. But I need help to do it on linux.

aymenim, 2019-12-31:
If I remember correctly you can set the locale and region  using `lockdownd_set_value` after activation and the pages won't show up  on the welcome screen.

jduranc, 2020-01-06:
Hi @aymenim ,

I already try doing that with no luck, I'm not sure if this could be done with the preboard service added by @nikias a few months ago.

closed 2021-02-03 by truonggiang0710

a-gunter, 2021-02-03:
`lockdownd_set_value` with the domain of `com.apple.international` will allow you to set Language / Locale.

Valid language & locale IDs: https://developer.apple.com/library/archive/documentation/MacOSX/Conceptual/BPInternational/LanguageandLocaleIDs/LanguageandLocaleIDs.html