libimobiledevice/libimobiledevice #693: Retry trust

mexmer, 2018-08-21:
anyone found a way to redisplay trust dialog after user denied trust?

i tried different stuff
regenerating systembuid
regenerating hostid
reinitializing whole communication
restarting usbmux
sending unpair request (although since it's not trusted, it's not paired ... this was just derpy attempt)

so far seems only way to get trust dialog to show again is disconnecting device from usb, looks like pc connection is remembered on device side, and trust cannot be reset until it's reconnected.

Kanong017, 2018-08-23:
I think there is a very strange way.

Reinstall the itunes driver service

mexmer, 2018-08-24:
disocnnecting phone is faster than reinstaling driver, so that's not the way. not to mention on windows for reinstaling driver you need admin privileges, on linux you don't have drivers to reinstall, they part of kernel. and no ... restarting usbmuxd (on linux), doesn't show trust dialog.

clavisound, 2018-08-29:
So your problem is that you don't want to reconnect the USB? You can unbund USB. (don't forget to first kill usbmuxd)

In my computer it worked.
`echo '2-1' > /sys/bus/usb/drivers/usb/unbind`
`killall usbmuxd` Maybe you need `killall -9 usbmuxd`
`echo '2-1' > /sys/bus/usb/drivers/usb/bind`
`usbmuxd -f`

type `lsusb` and `lsusb -t` to check your correct '2-1' numbers

mexmer, 2018-08-29:
hmm, didn't know about unbind, looks interesting, although stopping usbmuxd means admin ... but at least it doesn't require disconnect, so it's scriptable. now just need something like that for windows.

thanks

clavisound, 2018-08-29:
For Windows I suppose the equivalent is to disable the usb port from device manager?

I am pretty sure unbind / bind also needs admin :-)

But we have sudo for that.
example from sudoers `%disk	localhost = NOPASSWD: /usr/sbin/smartctl`

mexmer, 2018-08-30:
yes sudo will be way for script to do that.

as for windows, i think i tried that, but will try again. 

it's possible to disable/enable device in device manager programatically, tho' will need miniapp with eleveated privileges, it's doable.