libimobiledevice/libimobiledevice #695: Carrier Information

ankitrwt, 2018-08-29:
Hello, 
just want to know if there is any ioreg/ioregentry key available to get the current Carrier Information and also if there is any key to get carrier locked information of idevice, or any other source to get the same

Thanks

abdulzeedo, 2018-10-02:
Unfortunately, I am 99% sure that it is not possible to get carrier locked using this library. Your best bet would be to get this information directly from the IMIEIDB database owned by GSMA... 

But if there is another way, I would be happy to know it as well...

ankitrwt, 2019-01-16:
hello @abdulzeedo 
they only provide lost stolen mobile devices information,
is there any other way .. except GSMA and GSX
Any help will be appreciated
Thanks