libimobiledevice/libimobiledevice #698: Missing photo in DCIM/107IMPRT folder

jl6, 2018-09-02:
I use an Apple Lightning SD Card Reader to transfer photos from an SD card onto my iPhone 8+ (ios 11). The photos were taken with a Canon EOS 550D.

When I mount the iPhone on my Ubuntu 16.04 system using `idevicepair` and `ifuse`, I see the imported photos in the DCIM directory, in folders named `100IMPRT`, `101IMPRT`, `102IMPRT`, and so on - separate from the photos taken by the iPhone, which are in `100APPLE`, `101APPLE`, `102APPLE`, and so on.

In the folder named `107IMPRT`, one of my imported photos is missing: `IMG_2870.JPG`.

`IMG_2869.JPG` is there. `IMG_2871.JPG` is there. It's just this one that's missing.

It's not missing when I browse the Camera Roll using the Photos app on the iPhone.

It's not missing when I copy files directly off the original SD card onto my PC via a card reader.

Dropbox has also found the photo successfully and uploaded it automatically.

What could be causing `IMG_2870.JPG` in particular to not be available?

N.B. I am unable to query `idevicepair` to find out which version it is, but I installed it from a git clone on 9th January 2018.

mexmer, 2018-09-04:
if you have photo optimization enabled, it can be in cloud library assets. apple does weird stuff to photo since they added photo library.

jl6, 2018-09-04:
Confirming I do not have iCloud Photo Library enabled.