libimobiledevice/libimobiledevice #706: device_id -l  return 24 digit UDID on iPhone Xs(without - hyphen) , where ideviceinfo return 25digit UDID

omkarzunjurke, 2018-10-01:
On new release of iPhone Xs . device_id -l command return only 24 digit UDID, But in ideviceinfo udid is 25digit.The difference between this is only (-) hyphan. device_id -l does not print -(hyphen). In ideviceinfo  put put -(hyphen) is present.

Blefish, 2018-10-01:
https://github.com/libimobiledevice/libimobiledevice/commit/b34e3435c21d06b3e5a4e7b5246fb6ddb6641a9f

munimkazia, 2018-10-01:
You will need to update to the latest iTunes if on mac, or latest usbmuxd otherwise.

omkarzunjurke, 2018-10-04:
> @omkarzunjurke On which operating system is this? I find that for Windows, `idevice_id -l` will return the device ID without the dash, whereas _for the same device_ `idevice_id -l` on macOS will return the device ID with the dash.
> 
> It appears that the `ListDevices` Muxer command will return UDIDs including dashes on macOS and without dashes on Windows, but all other commands (such as `ReadPairRecord`) do expect the dash to be present.
> 
> I'd be interested in having this theory confirmed, so could you let me know which OS you are using?

Hello qmfredreik, i am running it on Linux machine.I got 24 character UDID on linux machine.

qmfrederik, 2018-10-04:
@omkarzunjurke That's interesting. When running `ideviceinfo -u [udid]`, do you need to include the hyphen or not?

Dantee296, 2018-10-04:
00008020-000D453C3E90003A
udid
IPHONE XS
@qmfrederik Windows

Dantee296, 2018-10-04:
@omkarzunjurke FYI , The UDID IN idevice_id is from usbmux  "usbmuxd_get_device_list"
while the ideviceinfo UDID is plist from  device Lockdown

mexmer, 2018-10-04:
@Dantee296 usbmuxd gets UDID from device USB setup packet (there is special usb packet to receive device identifier, if device supports that function), so if device sends identifier without hyphen, you know who to ask for this incosistency.

when i look in windows onto device instance path, it doesn't contain hyphen, so simple answer is ... blame apple for it.

munimkazia, 2018-10-04:
In my experience, I started getting the proper UDIDs with the hyphen after I upgraded the iTunes on my mac, because the native usbmuxd component got updated with it. For windows, I don't know if iTunes supplies the usbmuxd. But if it doesn't, you will have to see if the opensource usbmuxd (https://github.com/libimobiledevice/usbmuxd) has been updated to give device IDs with hyphen.

Blefish, 2018-10-04:
This causes #711 as well.

Dantee296, 2018-10-04:
well i tried and iphone xs and both ideviceid and ideviceinfo showing same with hypen

mexmer, 2018-10-04:
@Dantee296 that UDID from system report confirms what i said about USB communication. device returns identifier without hyphen when directly asked trough usb. but when you talk with lockdown service (which is primary channel), you get id with hyphen.

usbmuxd can get udid from 2 sources ...
1) direct USB request
2) preflight communication with lockdown service

technically for pairing matters what lockdown service says, not what USB says, if USB identification request will say "thisisblueiphone" instead of UDID .... well you know, what i mean.

usbidentification matters only when lockdown service is not accessible (dfu mode for example), so you can distinguish between multiple devices.

although on windows, like on mac, usbmuxd communicates with apple mux service, which is installed with itunes ... so might be, that on windows, it has not been updated yet.

omkarzunjurke, 2018-10-04:
> @omkarzunjurke That's interesting. When running `ideviceinfo -u [udid]`, do you need to include the hyphen or not?

@qmfrederik  yes that is interesting i did not need to enter hyphen (-) while using ideviceinfo -u .if u enter hyphen (-) it did not work.

nikias, 2018-10-05:
ok so the original usbmuxd on macOS is definitely adding the hyphen:
```__text:000000010000A9D3                 mov     rdi, rbx        ; char *
__text:000000010000A9D6                 call    _strlen
__text:000000010000A9DB                 cmp     eax, 18h
__text:000000010000A9DE                 jnz     short loc_10000AA15
__text:000000010000A9E0                 mov     esi, 18h
__text:000000010000A9E5                 mov     rdi, rbx
__text:000000010000A9E8                 call    sub_100005044
__text:000000010000A9ED                 test    al, al
__text:000000010000A9EF                 jz      short loc_10000AA15
__text:000000010000A9F1                 mov     al, [r12+40h]
__text:000000010000A9F6                 mov     [r12+41h], al
__text:000000010000A9FB                 mov     rax, [r12+30h]
__text:000000010000AA00                 mov     rcx, [r12+38h]
__text:000000010000AA05                 mov     [r12+39h], rcx
__text:000000010000AA0A                 mov     [r12+31h], rax
__text:000000010000AA0F                 mov     byte ptr [r12+30h], 2Dh ; '-'```

dnsBlah, 2018-10-05:
HI all,

I have an XS Max here, idevice_id -l shows 24 characters, without the hyphen.
When I paired the device ideviceinfo shows
"ERROR: Could not connect to lockdownd, error code -2"

Dantee296, 2018-10-05:
@dnsBlah  apply # @[https://github.com/libimobiledevice/libimobiledevice/commit/b34e3435c21d06b3e5a4e7b5246fb6ddb6641a9f](url)

Dantee296, 2018-10-05:
@dnsBlah  b34e343

dnsBlah, 2018-10-05:
@Dantee296 I just downloaded all the latest libraries a 1.5 hours ago.
I got these updates.

idevice_id -l shows no hypen.
idevicepair pairs fine without hypen
but idevieinfo gives me error -2

qmfrederik, 2018-10-05:
@dnsBlah Is this on windows? Which binaries did you use?

qmfrederik, 2018-10-05:
@nikias Here's what I've observed:

| OS                      | `idevice_id -l` includes hyphen?     | `ideviceinfo -u` [with hyphen] | `ideviceinfo -u` [no hyphen]` |
|-------------------|--------------------------------------|------------------------------------|-----------|
| macOS   | Yes | Works | Error  |
| Windows | No | Works | Error |

This is Windows with iTunes 12.9.0.167, Apple Mobile Device Support 12.0.0.1039

It's annoying that the multiplexor on Windows returns the UDID without the hyphen but then requires the hyphen. 

A workaround would be make `libusbmuxd` ensure the UDID includes a hyphen, like https://github.com/libimobiledevice-win32/libusbmuxd/commit/04e20e2072dcb11a000399d97fd871c2d3dbcd57.

dnsBlah, 2018-10-05:
Hi @qmfrederik,
thanks for your reply! I'm running on OSX

I did manually download all the libraries using git clone and installed them.
- brew install libusb
- libusbmuxd - autogen
- usbmuxd - autogen
- libimobiledevice - autogen --disable-openssl

The idevice_id -l does not show the hyphen

nikias, 2018-10-05:
@dnsBlah why do you install usbmuxd on macOS? If you used the latest usbmuxd from git, `idevice_id 
 -l` WILL show the hyphen...

Mradr, 2018-10-05:
I seem to be getting an error on Windows even with hyphen using the ideviceinfo -u for iPhone X. 8 and lower seem fine.

Mradr, 2018-10-07:
Rename device was also displaying this issue - just wanted to make sure it was on the list as well of fix.

dnsBlah, 2018-10-08:
@nikias There is something more behind it then only usbmuxd I guess.
I've made a clean install
```
- install brew
brew install automake
brew install pkg-config
brew install gnutls
brew install libgcrypt

git clone https://github.com/libimobiledevice/libusbmuxd.git
cd libusbmuxd
./autogen
make
sudo make install

cd ..
git clone https://github.com/libimobiledevice/libimobiledevice.git
cd libimobiledevice
./autogen.sh --disable-openssl
make
sudo make install
```

After the above idevice_id still does not show the hyphen
I've copied the compiled tools to an Mojave machine, and there it did work.

```
cd ..
git clone https://github.com/libimobiledevice/usbmuxd.git
cd usbmuxd
./autogen.sh
make
sudo make install
```
Still same behaviour, not showing hypen, yet idevicepair pairs succesfully but ideviceinfo is giving error -2

Any advice, besides upgrading the OS for all our devices ? 👍

qmfrederik, 2018-10-08:
@dnsBlah You don't need to install usbmuxd on macOS. I don't think usbmuxd registers itself as a launch daemon on macOS, so installing it doesn't really do anything.

What you are probably seeing is that on High Sierra, `idevice_id -l` does not include the hyphen (but the tools do require it), whereas this is fixed on Mojave. The difference is in the version of usbmuxd that ships with macOS (not the one you compile from libimobiledevice).

So it looks like https://github.com/libimobiledevice-win32/libusbmuxd/commit/04e20e2072dcb11a000399d97fd871c2d3dbcd57 would be relevant on macOS versions earlier than Mojave, too.

dnsBlah, 2018-10-08:
> @dnsBlah You don't need to install usbmuxd on macOS. I don't think usbmuxd registers itself as a launch daemon on macOS, so installing it doesn't really do anything.
> 
> What you are probably seeing is that on High Sierra, `idevice_id -l` does not include the hyphen (but the tools do require it), whereas this is fixed on Mojave. The difference is in the version of usbmuxd that ships with macOS (not the one you compile from libimobiledevice).
> 
> So it looks like [libimobiledevice-win32/libusbmuxd@04e20e2](https://github.com/libimobiledevice-win32/libusbmuxd/commit/04e20e2072dcb11a000399d97fd871c2d3dbcd57) would be relevant on macOS versions earlier than Mojave, too.

Hi Frederik,

it was just a test.
I can up on this error on my own hook too.
See: https://github.com/libimobiledevice/libimobiledevice/issues/711#issuecomment-427668264
libusbmuxd is not cleaning the serial/udid on alphanumeric.

I've installed xCode on the high sierra, and now it works. I need to find out why and then I can role out the new version. :)

dnsBlah, 2018-10-09:
I had to actually install iTunes 12.0.8 or higher to get rid of the error -2
Even tho I disable the license agreement when using my application.

I'm using these libraries natively on OSX by using export DYLD_LIBRARY_PATH to the folder of the used tools and add the required library files.

Any advice on using the libraries in a more native way? I'd expect that the iTunes update did update my local usbmuxd. I've tried to use usbmuxd in the dyld folder as well, but it's a daemon which will re-run when killed, plus it's SIP protected.