libimobiledevice/libimobiledevice #707: Configure error

plinytheelder, 2018-10-02:
trying to set this up again on another computer, but when I install libimobiledevice I get this error:
configure: error: Package requirements (libusbmuxd >= 1.1.0) were not met:

Requested 'libusbmuxd >= 1.1.0' but version of libusbmuxd is 1.0.10

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libusbmuxd_CFLAGS
and libusbmuxd_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

I looked at your GitHub and commit f1ee0fd8b2b1dd65c64f5a67e208efc708295caa recently changed libusbmuxd version from 1.0.9 to 1.1.0. But when I check your website, only 1.0.10 is available. so I don't think it's me...

plinytheelder, 2018-10-02:
nope. same error. I've installed this on few instances the past few weeks and haven't had a problem up until now. my friend reported the same.

mexmer, 2018-10-02:
first uninstall any libusbmux and libusbmux-dev you have already

then download head revision from git, it will generate version 1.1.0, of course you need to do clean build, then install compiled version

if configure still cannot find libusmux, you can manually set path to libusbmux compiled from source

plinytheelder, 2018-10-02:
git pulled libusbmuxd and ran set-up, still didn't work. 
but `brew install --HEAD usbmuxd` installed it correctly.