libimobiledevice/libimobiledevice #709: Error: undefined reference to `usbmuxd_save_pair_record_with_device_id'

mcplectrum, 2018-10-03:
Hi,
I used this script: http://pastebin.com/6u0TEhjz to install the latest versions and running *./autogen.sh --enable-debug-code* results in:
```
../src/.libs/libimobiledevice.so: undefined reference to `usbmuxd_save_pair_record_with_device_id'
collect2: error: ld returned 1 exit status
Makefile:653: recipe for target 'idevice_id' failed
```
i did a grep on *usbmuxd_save_pair_record_with_device_id*, which results in:
./libusbmuxd-master/include/usbmuxd.h:221:int usbmuxd_save_pair_record_with_device_id(const char* record_id, uint32_t device_id, const char *record_data, uint32_t record_size);

mcplectrum, 2018-10-03:
Solution:
1. $ sudo vim /etc/ld.so.conf
#file content:
/usr/local/lib
2. sudo ldconfig
3. make && sudo make install

dnsBlah, 2018-10-05:
I get the same when building on OSX 10.14 Mojave

beret, 2018-10-05:
I had this same issue with the latest builds. Once I rolled back to fb71aeef10488ed7b0e60a1c8a553193301428c0 and was able to compile without issue.

Unfortunately the workaround you mentioned didn't work for me. My (updated from git) usbmuxd headers are in /usr/include. I applied that and /usr/local/lib, with no change in the issue.

Arch x64, linux-lts 4.14.69, gcc 8.2.1

```
Making all in tools
make[2]: Entering directory '/home/user/build/libimobiledevice-git/src/libimobiledevice/tools'
  CCLD     idevice_id
  CCLD     ideviceinfo
  CCLD     idevicename
  CCLD     idevicepair
/usr/bin/ld: ../src/.libs/libimobiledevice.so: undefined reference to `usbmuxd_save_pair_record_with_device_id'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:697: idevicename] Error 1
make[2]: *** Waiting for unfinished jobs....
/usr/bin/ld: ../common/.libs/libinternalcommon.a(userpref.o): in function `userpref_save_pair_record':
userpref.c:(.text+0x37f): undefined reference to `usbmuxd_save_pair_record_with_device_id'
collect2: error: ld returned 1 exit status
/usr/bin/ld: ../src/.libs/libimobiledevice.so: undefined reference to `usbmuxd_save_pair_record_with_device_id'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:705: idevicepair] Error 1
make[2]: *** [Makefile:653: idevice_id] Error 1
/usr/bin/ld: ../src/.libs/libimobiledevice.so: undefined reference to `usbmuxd_save_pair_record_with_device_id'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:693: ideviceinfo] Error 1
make[2]: Leaving directory '/home/user/build/libimobiledevice-git/src/libimobiledevice/tools'
make[1]: *** [Makefile:455: all-recursive] Error 1
make[1]: Leaving directory '/home/user/build/libimobiledevice-git/src/libimobiledevice'
make: *** [Makefile:385: all] Error 2
```

b-m-f, 2018-11-04:
Try removing the package that is provided by the distro of `libusbmuxd`. Then compiling from https://github.com/libimobiledevice/libusbmuxd and using the solution  from @mcplectrum should work.

ethanas, 2018-11-26:
I get the same when building on raspberry pi 3B+（Raspbian GNU/Linux 9.1 (stretch)）

ethanas, 2018-11-27:
I also rolled back to fb71aee ，that is ok

cayenne3, 2018-12-28:
Same here. Raspberry Pi 3B+ running Rasbian (stretch).  @mcplectrum 's workaround did not work for me either but rolling back to [https://github.com/libimobiledevice/libimobiledevice/commit/fb71aeef10488ed7b0e60a1c8a553193301428c0] worked.

jbycdmys, 2019-01-11:
Thanks tinwasp! After updating Arch to 4.20.0, I had multiple issues with xfce, including upower failing to start. Downgrading libimobiledevice solved my issues. Thanks

rajpancha1, 2019-01-18:
Hi All:

I am new to the libimobiledevice forum.

Like Ethanas, I am also trying to install libimobiledevice suite on Raspberry PI 3B+ (Stretch Lite with openbox). Linux SSbase 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux.

I am also getting the same error:
-----------
  CC       idevice_id-idevice_id.o
  CCLD     idevice_id
../src/.libs/libimobiledevice.so: undefined reference to `usbmuxd_save_pair_record_with_device_id'
collect2: error: ld returned 1 exit status

I am following steps from:  raspbian-install.sh on https://gist.github.com/kfatehi/8922430#file-raspbian-install-sh-L13.

Can someone please guide me through to resolve the issue?
Should I use the branch https://github.com/libimobiledevice/libimobiledevice/tree/1.1.7 for all the related packages?

I greatly appreciate your help.

Thanks

nikias, 2019-01-18:
You seem to have an older libusbmuxd package installed. Make sure to remove that.

nikias, 2019-01-18:
In that case you somehow managed to install an older version of libusbmuxd.

rajpancha1, 2019-01-18:
So how do I install the right one?

I just followed the raspbian-install.sh script  on
https://gist.github.com/kfatehi/8922430#file-raspbian-install-sh-L13.

Appreciate your prompt reply. I need to install the suite so that when a user connects the iOS device on the RPI3 B+, it gets notified with information about the hosted iOS device, at least for now.

TitanFighter, 2019-01-18:
Just for those who does not know what to does it mean:
> I had this same issue with the latest builds. Once I rolled back to [fb71aee](https://github.com/libimobiledevice/libimobiledevice/commit/fb71aeef10488ed7b0e60a1c8a553193301428c0) I was able to compile without issue.

```
$ git clone https://github.com/libimobiledevice/libimobiledevice.git
$ cd libimobiledevice
$ git reset --hard fb71aeef10488ed7b0e60a1c8a553193301428c0
```

nikias, 2019-01-18:
@TitanFighter that is not the right approach. The symbol it can't find was added to libusbmuxd on September 15 with commit https://github.com/libimobiledevice/libusbmuxd/commit/07a493a570ddef69f5c424795345813fc9c32d6d and libimobiledevice was updated to use that on September 29 with commit https://github.com/libimobiledevice/libimobiledevice/commit/f1ee0fd8b2b1dd65c64f5a67e208efc708295caa.
I am using latest git head and it works fine.

nikias, 2019-01-18:
I just installed it on my raspberry pi 3 with the following commands:
```

nikias, 2019-01-18:
@rajpancha1 check if you have any other libusbmuxd leftovers:
```bash
$ find /usr -name "libusbmuxd*"
/usr/local/lib/libusbmuxd.so
/usr/local/lib/libusbmuxd.so.4
/usr/local/lib/libusbmuxd.a
/usr/local/lib/pkgconfig/libusbmuxd.pc
/usr/local/lib/libusbmuxd.so.4.1.0
/usr/local/lib/libusbmuxd.la
```
That is my output. Just one library there.

rajpancha1, 2019-01-18:
Thank you TitanFighter and Nikias.

Following  TitanFighter's suggestion I rebuilt the libimobiledevice.

Then following Nikias suggestion, here's the output:
$ ls -ltra `find /usr -name "libusbmuxd*"`
-rw-r--r-- 1 root root      86 Jun  4  2016 /usr/share/lintian/overrides/libusbmuxd4
lrwxrwxrwx 1 root root      19 Jun  9  2016 /usr/lib/arm-linux-gnueabihf/libusbmuxd.so.4 -> libusbmuxd.so.4.0.0
-rw-r--r-- 1 root root   30316 Jun  9  2016 /usr/lib/arm-linux-gnueabihf/libusbmuxd.so.4.0.0
-rwxr-xr-x 1 root staff  96812 Jan 18 08:25 /usr/local/lib/libusbmuxd.so.4.1.0
lrwxrwxrwx 1 root staff     19 Jan 18 08:25 /usr/local/lib/libusbmuxd.so.4 -> libusbmuxd.so.4.1.0
lrwxrwxrwx 1 root staff     19 Jan 18 08:25 /usr/local/lib/libusbmuxd.so -> libusbmuxd.so.4.1.0
-rwxr-xr-x 1 root staff   1015 Jan 18 08:25 /usr/local/lib/libusbmuxd.la
-rw-r--r-- 1 root staff 116154 Jan 18 08:25 /usr/local/lib/libusbmuxd.a
-rw-r--r-- 1 root staff    276 Jan 18 08:25 /usr/local/lib/pkgconfig/libusbmuxd.pc

/usr/share/doc/libusbmuxd4:
total 64
-rw-r--r--    1 root root  2898 Jun  4  2016 copyright
-rw-r--r--    1 root root   516 Jun  4  2016 changelog.Debian.gz
drwxr-xr-x    2 root root  4096 Aug 18 13:23 .
drwxr-xr-x 1351 root root 49152 Jan 18 09:22 ..

TitanFighter, 2019-01-18:
@rajpancha1 Thanks for step-by-step commands. Also in my case the command:
`ideviceinfo -k ProductVersion`
raised issue
`ERROR: Could not connect to lockdownd, error code -2`.

Found the solution [here](https://github.com/libimobiledevice/libimobiledevice/issues/365#issuecomment-334431298), ie `libimobiledevice` must be generated like this `./autogen.sh --enable-openssl`

nikias, 2019-01-18:
@rajpancha1 as you can see, you have an older version from 2016:
```
lrwxrwxrwx 1 root root     19 Jun 9 2016 /usr/lib/arm-linux-gnueabihf/libusbmuxd.so.4 -> libusbmuxd.so.4.0.0
-rw-r--r-- 1 root root  30316 Jun 9 2016 /usr/lib/arm-linux-gnueabihf/libusbmuxd.so.4.0.0
```
So when linking libimobiledevice it will use that one. Also there's
```
/usr/share/lintian/overrides/libusbmuxd4
```
Which denotes that it selected another package as default package.
Try removing the libusbmuxd4 and the other packages:
```
sudo apt-get remove libplist3 libusbmuxd4 libimobiledevice6
```
If it still says package not installed, remove the files manually:
```
sudo rm /usr/lib/arm-linux-gnueabihf/libusbmuxd.so.4 /usr/lib/arm-linux-gnueabihf/libusbmuxd.so.4.0.0
```
Also remove the lintian file:
```
sudo rm /usr/share/lintian/overrides/libusbmuxd4
```
Good luck.

nikias, 2019-01-18:
@TitanFighter there is no need to pass `--enable-openssl` if you have `libssl-dev` installed since it will select OpenSSL by default (unless you configured it previously to use GnuTLS). The same lockdown error is also thrown if you didn't trust the computer on the device yet, btw.

rajpancha1, 2019-01-18:
I just started restoring the flash from my pre-libimobiledevice journey.
I will restart clean install based on guidance from both of you.
I had install libssl-dev.

I'll let you know soon.
Thanks

rajpancha1, 2019-01-19:
Great! It works!
Thank TitanFighter and Nikias for your help.
One more question. How do I create a notification when any device is connected?

sudo service usbmuxd start
sudo service usbmuxd status
$ sudo service usbmuxd status
? usbmuxd.service - Socket daemon for the usbmux protocol used by Apple devices
   Loaded: loaded (/lib/systemd/system/usbmuxd.service; static; vendor preset: enabled)
   Active: active (running) since Fri 2019-01-18 19:56:01 CST; 10min ago
     Docs: man:usbmuxd(8)
 Main PID: 2220 (usbmuxd)
   CGroup: /system.slice/usbmuxd.service
           mq2220 /usr/local/sbin/usbmuxd --user usbmux --systemd
....
....
$ idevice_id -l
11a00339c2da93672ae1b56c393a49921b4e109d

$ ideviceinfo -k ProductVersion
9.3.5
$ idevicename
Darshanas iPad

nikias, 2019-01-21:
@rajpancha1 I never attached mouse or keyboard to the RPi but I somehow doubt this is related in any way. Is this happening also if no iOS device is plugged in? If you check the process list for usbmuxd:
```
ps axgu |grep usbmuxd
```
If it doesn't show anything (except for a line with `grep usbmuxd`) then it is definitely not because of libimobiledevice since nothing is running that could interfere with HID.