libimobiledevice/libimobiledevice #714: [idevicebackup2] Restore to iphone 8 ios 12 worked but made boot loop

alexngco, 2018-10-05:
Hi.
I compiled the master branch using msys on window 10 (using the option --without-cython). Everything works fine.
I tried to perform a backup using `idevicebackup2 backup --full path` from an Iphone 5 ios 10. It worked.
Then I tried to restore this package to another Iphone 8 ios 12, the restore also worked well by executing `idevicebackup2 restore -u ## -s ## --system --settings --reboot path` (I got `Restore successful`) .
However my phone is stuck on boot loop just after that. 

Did iOS 12 break something on the restore ?

Blefish, 2018-10-08:
I haven't faced this problem, I suggest creating backup from a newer iOS version if you need to restore it to a different device.

alexngco, 2018-10-08:
I would like to. But I got some use cases that need to cover it.
I noticed that removing `--settings` fix the boot loop issue.