libimobiledevice/libimobiledevice #717: ERROR: Could not connect to lockdownd, error code -3

MastroGeppetto14, 2018-10-08:
Raspberrry pi3 Stretch

I can do "idevice pair" and unpair, but just this. Can anyone help me?

aeschenko, 2018-10-09:
same for me
as a long-time lib user (we have inhouse-framework based on the lib), I tried all the tricks those worked before, like this: https://github.com/libimobiledevice/ideviceinstaller/issues/48#issuecomment-292186533
now these tricks do not work for me anymore

schichagov, 2018-10-09:
Also have faced with "ERROR: Could not connect to lockdownd, error code -3" and "-21" after reinstall libimobiledevice

MastroGeppetto14, 2018-10-10:
@aeschenko, @schichagov  Hi sir,

i tried everything (chmod lockdown.c, reinstall all, update raspbian, try older release, delete files from urs/local/lib). Today i tried change iphone.  Last days i used an iphone 6 with ios12, today i try with iphone 4 with ios 7 and iphone5 with ios 10. 
These work finally, so this problem has create from ios 12. 

To resolve this problem, we have to wait an update from libimobile or use an device with ios 11 or lower.

Greetings

(sorry for my bad english)

MastroGeppetto14, 2018-10-11:
> @MastroGeppetto14 Are you facing this issue on macOS Mojave or macOS High Sierra? For me, it was working fine on macOS High Sierra. I am facing this issue ever since I upgraded to macOS Mojave.

Nope  i'm using Ubuntu / Raspbian stretch
both give me problems

MastroGeppetto14, 2018-10-15:
i solved the problems abount ubuntu 16 (i installed ubuntu mate 16 on raspberry)
Using this guide:
https://gist.github.com/samrocketman/70dff6ebb18004fc37dc5e33c259a0fc

Ios 12 works fine   even with patches 12.0.1/2/3

aeschenko, 2018-10-17:
finally works after

brew uninstall --ignore-dependencies libimobiledevice
brew uninstall --ignore-dependencies ideviceinstaller
brew uninstall --ignore-dependencies usbmuxd
sudo rm /var/db/lockdown/*
brew install --HEAD usbmuxd
brew unlink usbmuxd
brew link usbmuxd
brew install --HEAD libimobiledevice
brew install --HEAD ideviceinstaller

closed 2018-11-08 by MastroGeppetto14

AngleZhou, 2019-03-15:
@aeschenko Thanks. It also works for macOS Mojave.

CYsuncheng, 2019-05-15:
> finally works after
>
> brew uninstall --ignore-dependencies libimobiledevice
> brew uninstall --ignore-dependencies ideviceinstaller
> brew uninstall --ignore-dependencies usbmuxd
> sudo rm [...]

thanks a lot

mattsears18, 2019-06-30:
I also had to unplug my iPhone before running the commands listed by @aeschenko. Thanks! I spent waaaaay too much time on this.

NSExceptional, 2020-09-15:
This did not work for me. I get a few errors, some like `No such keg: /usr/local/Cellar/libusbmuxd` when trying to `(un)link usbmuxd`

PengJiang520, 2021-12-06:
brew install --HEAD usbmuxd
Running `brew update --preinstall`...
Warning: Use libusbmuxd instead of deprecated usbmuxd
Warning: Use libusbmuxd instead of deprecated usbmuxd
Warning: Use libusbmuxd instead of deprecated usbmuxd
==> Downloading https://mirrors.ustc.edu.cn/homebrew-bottles/libplist-2.2.0.montere
######################################################################## 100.0%
==> Cloning https://github.com/libimobiledevice/libusbmuxd.git
Updating /Users/jiangpeng/Library/Caches/Homebrew/libusbmuxd--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 2ec5354 [github-actions] macOS: build for arm64 too
==> Installing dependencies for libusbmuxd: libplist
==> Installing libusbmuxd dependency: libplist
==> Pouring libplist-2.2.0.monterey.bottle.tar.gz
🍺  /usr/local/Cellar/libplist/2.2.0: 31 files, 492.2KB
==> Installing libusbmuxd --HEAD
==> ./autogen.sh
Last 15 lines from /Users/jiangpeng/Library/Logs/Homebrew/libusbmuxd/01.autogen.sh:
checking whether to build static libraries... yes
checking for pkg-config... /usr/local/Homebrew/Library/Homebrew/shims/mac/super/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libplist-2.0 >= 2.2.0... yes
checking for libimobiledevice-glue-1.0 >= 1.0.0... no
configure: error: Package requirements (libimobiledevice-glue-1.0 >= 1.0.0) were not met:

No package 'libimobiledevice-glue-1.0' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables limd_glue_CFLAGS
and limd_glue_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

READ THIS: https://docs.brew.sh/Troubleshooting

Please create pull requests instead of asking for help on Homebrew's GitHub,
Twitter or any other official channels