libimobiledevice/libimobiledevice #720: iTunes Wifi Sync

Soutcast, 2018-10-09:
There seems to be a bit of uncertainty about whether or not you can toggle iTunes Wifi Sync through libimobiledevice or not. As I would like to resolve this confusion I ask. Does anyone know whether you can actually toggle iTunes Wifi Sync through libimobiledevice? If so, could you please post the steps taken in the comments below.

If this is not the case then I would also like to know if it is possible to toggle the setting through a terminal command or script of some sort. Again, if there is a script or command which can toggle iTunes Wifi Sync, could you please post the code in the comments below.

Through the hours of web trawling I have done, I have been unable to determine whether iTunes Wifi Sync can be toggled through libimobiledevice and if there is a script or command to toggle iTunes Wifi Sync. So, if anyone knows anything about the two issues I have discussed I urge you, please help me out here and let me know.

nikias, 2018-10-13:
I am working on supporting wifi sync properly, and will add a tool for it aswell.

Soutcast, 2018-10-13:
Just a question, if libimobiledevice doesn't have proper support for itunes wifi sync then how do symantec use libimobiledevice commands remotely via itunes wifi sync?
[demo shown here](https://www.youtube.com/watch?v=3TWCSfBhtss)

Soutcast, 2018-10-14:
Hi @nikias, I also noticed that you committed a change to the libusbmuxd repo which states that it adds proper support for wifi sync devices reported by usbmuxd. Is this related to your work to support wifi sync communication in libimobiledevice?

mexmer, 2018-10-15:
mind on OS X as well as on Windows, libimobile is communicating with apple provided mux service (apple mobile service), which handles USB and wifi connection likewise, that's why also people observe duplicate UDID on those OSes, when wifi sync for device is enabled.

apple device mux communicates with devices using TCP protocol (which is sent either over USB or wifi ... whatever connection is preferred).

so yes,, when you can communicate with mux, and mux is already authenticated with device (trusted,paired), then you can do, what is described in that video, but that was possible since wifi sync first came out. there was never explicit consent for wifi sync. i don't remember which was first ios and itunes version with wifisync, but i believe it goes quite far back (might be even ios 4/itunes 7). really not sure, when was first time, that "wifi sync" option appeared in itunes.

but still for wifi sync to work, device first need to be connected trough USB once and of course needs confirm trust, and you can only enable wifi sync, when device is connected trough usb.

i'm not sure if it's possible to disable wifi sync from iOS UI tho'

Soutcast, 2018-10-15:
Hi @mexmer, I understand what you are saying and thanks for the reply. If you have read my previous comment to @nikias about the Symantec trustjacking attack then you should know that I have been having trouble communicating over iTunes wifi sync. Here is the rundown on my current situation.

I am trying to recreate the trustjacking attack presented by Symantec at RSA 2018 but I have been unable to execute libimobiledevice commands remotely through iTunes wifi sync.

When I have a physical USB connection between my ios device and the computer all the commands work perfectly but when I enable itunes wifi sync and disconnect the cable I can no longer contact to the device.

However, when I run `idevice_id -l` over wifi it displays the uuid of my ios device but any other commands just display errors such as `ERROR: Could not connect to device` or `No device found, is it plugged in?`.

Any suggestions on how I can resolve this problem?

Soutcast, 2018-10-16:
Its turns out that my version of libimobiledevice was corrupt as well as the dependencies. I fixed this by completely reinstalling libimobiledevice and its dependencies as well as removing all previous data stored by libimobiledevice.

ila08, 2019-02-20:
with reference to URL : https://www.symantec.com/blogs/feature-stories/ios-trustjacking-dangerous-new-ios-vulnerability
where the python command required to run (demonstration)?  Are they using any emulator or malicious software ?

Soutcast, 2019-02-20:
Hi @ila08,
With regards to the trustjacking demo the researchers created their own python scripts which interact with the libimobiledevice library to perform the actions seen in the demo. These scripts are not affiliated with the libimobiledevice library.

ila08, 2019-02-20:
Is that possible to get access to it ?? I am trying to exploit trust jacking vulnerability. Could you please help on the same

Soutcast, 2019-02-20:
@ila08,
The researchers have not released their script publicly but with some python knowledge it wouldn't be to difficult to recreate the scripts to perform similar actions

ila08, 2019-02-20:
Could you please suggest the python modules that are required to create the scripts w.r.t trustjacking ?

Soutcast, 2019-02-20:
@ila08, It depends on which script you are trying to recreate. For example, if you wanted to recreate video jacking script you would need modules such as Tkinter and PIL for the image display and io for stream handling.

ila08, 2019-02-20:
OK that was helpful, thank you !!

Soutcast, 2019-02-20:
@ila08, Just out of interest which part of the trustjacking attack would you like to replicate?

ila08, 2019-02-21:
want to start with streaming the screen

chengaomin, 2019-05-01:
@Soutcast @nikias 

> However, when I run idevice_id -l over wifi it displays the uuid of my ios device but any other commands just display errors such as ERROR: Could not connect to device or No device found, is it plugged in?.

> Its turns out that my version of libimobiledevice was corrupt as well as the dependencies. I fixed this by completely reinstalling libimobiledevice and its dependencies as well as removing all previous data stored by libimobiledevice.

I had the same problem.
I try to uninstall libimobiledevice  and reinstall ，but the problem is still.

macOS : 10.14.4
iPhone 7P : 12.1.4
iTunes:12.9.4.94 ( Wifi Sync )

Fidetro, 2020-05-04:
Hi,@nikias
Have any progress about toggle iTunes Wifi Sync through libimobiledevice? :)

didix21, 2021-04-19:
Hello I'm really interested WiFi sync feature, are there any updates on this issue ? Thanks