libimobiledevice/libimobiledevice #721: ERROR: Could not connect to lockdownd, error code -21

aliimran, 2018-10-10:
I am facing this issue while using idevicescreenshot command on macOS Mojave. It was working fine for me on macOS High Sierra.

MastroGeppetto14, 2018-10-11:
In my opinion the cause is update, like ios 12

aliimran, 2018-10-11:
Maybe it is libimobiledevice update 1.2.0_3 that has caused this issue. Is there any way I can install the previous version using homebrew?

MastroGeppetto14, 2018-10-11:
This issue confirm my opinion about update of mac 
https://github.com/libimobiledevice/libimobiledevice/issues/722

ha100, 2018-11-25:
same problem here. ios12 macOS High Sierra 10.13.6 running `ideviceinfo -d`

```
03:09:34 lockdown.c:654 lockdownd_client_new(): device udid: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
03:09:34 lockdown.c:384 lockdownd_query_type(): called
03:09:34 property_list_service.c:127 internal_plist_send(): sending 284 bytes
03:09:34 service.c:140 service_send(): sending 4 bytes
03:09:34 service.c:140 service_send(): sending 284 bytes
03:09:34 property_list_service.c:132 internal_plist_send(): sent 284 bytes
03:09:34 property_list_service.c:133 internal_plist_send(): printing 284 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>Request</key>
	<string>QueryType</string>
</dict>
</plist>
03:09:34 property_list_service.c:191 internal_plist_receive_timeout(): initial read=4
03:09:34 property_list_service.c:200 internal_plist_receive_timeout(): 297 bytes following
03:09:34 property_list_service.c:213 internal_plist_receive_timeout(): received 297 bytes
03:09:34 property_list_service.c:239 internal_plist_receive_timeout(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Request</key>
	<string>QueryType</string>
	<key>Type</key>
	<string>com.apple.mobile.lockdown</string>
</dict>
</plist>
03:09:34 lockdown.c:400 lockdownd_query_type(): success with type com.apple.mobile.lockdown
03:09:34 userpref.c:170 userpref_get_config_dir(): initialized config_dir to /var/db/lockdown
03:09:34 property_list_service.c:127 internal_plist_send(): sending 333 bytes
03:09:34 service.c:140 service_send(): sending 4 bytes
03:09:34 service.c:140 service_send(): sending 333 bytes
03:09:34 property_list_service.c:132 internal_plist_send(): sent 333 bytes
03:09:34 property_list_service.c:133 internal_plist_send(): printing 333 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>Key</key>
	<string>DevicePublicKey</string>
	<key>Request</key>
	<string>GetValue</string>
</dict>
</plist>
03:09:34 property_list_service.c:191 internal_plist_receive_timeout(): initial read=4
03:09:34 property_list_service.c:200 internal_plist_receive_timeout(): 335 bytes following
03:09:34 property_list_service.c:213 internal_plist_receive_timeout(): received 335 bytes
03:09:34 property_list_service.c:239 internal_plist_receive_timeout(): printing 335 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Error</key>
	<string>GetProhibited</string>
	<key>Key</key>
	<string>DevicePublicKey</string>
	<key>Request</key>
	<string>GetValue</string>
</dict>
</plist>
03:09:34 lockdown.c:180 lockdown_check_result(): ERROR: GetProhibited
03:09:34 lockdown.c:791 pair_record_generate(): device refused to send public key.
ERROR: Could not connect to lockdownd, error code -13
```

mexmer, 2018-11-25:
Seems like device policy on your phone is blocking connection with computer.

mleonhard, 2019-03-05:
I fixed this problem by installing the latest committed code for ideviceinstaller and its dependencies.  I'm using Macports so I needed to first uninstall libusbmuxd, libimobiledevice, and ideviceinstaller.  Then I installed the '-devel' versions of those modules.  Now I have:

$ sudo port installed |grep 'dev'
  ideviceinstaller-devel @20181001_0 (active)
  libimobiledevice-devel @20190217_0 (active)
  libusbmuxd-devel @20190119_0 (active)
$ ideviceinfo |grep DeviceName
DeviceName: Michael’s iPhone
$

olaide-ekolere, 2019-07-22:
I just found out this error occurs when a friend is charging his iphone on my mac, immediately i disconnect it it starts working, strange right.

harmjanblok, 2019-07-22:
I had a similar error `Failed to connect to lockdownd service on the device.`, using Mojave (10.14.4) and iOS 12.3.1. After installing latest versions for `libimobiledevice` and `usbmuxd` I was able to mount my iPhone. See #775 for detailed instructions on how to install latest (`HEAD`) version using homebrew.

```
brew unlink usbmuxd; brew install --HEAD usbmuxd
brew unlink libimobiledevice; brew install --HEAD libimobiledevice
```

waelsaad, 2019-08-03:
> brew unlink usbmuxd; brew install --HEAD usbmuxd
> brew unlink libimobiledevice; brew install --HEAD libimobiledevice

I tried so many things on **Mojave 10.14.6** but I keep getting this error:

01:09:42 lockdown.c:180 lockdown_check_result(): ERROR: GetProhibited
01:09:42 lockdown.c:791 pair_record_generate(): device refused to send public key.
ERROR: Could not connect to lockdownd, error code -13

feudadrien, 2019-08-25:
Disconnect your iPhone Device Or Android Simulator then reconnect again. Just worked for me

parth22, 2019-08-30:
> I just found out this error occurs when a friend is charging his iphone on my mac, immediately i disconnect it it starts working, strange right.

yes that's right. it's really strange.

mtshare, 2019-09-05:
@harmjanblok thanks, it works!

DebugTheCode, 2019-09-06:
> I just found out this error occurs when a friend is charging his iphone on my mac, immediately i disconnect it it starts working, strange right.

Thanks a lot! Disconnected my iPhone because it was also charging as Android phone on my Mac.

mjhansen3, 2019-09-10:
> I just found out this error occurs when a friend is charging his iphone on my mac, immediately i disconnect it it starts working, strange right.

Strangely it worked ... my iPhone was plugged in charging but I was building on the simulator. It worked once I unplugged it.

amitkanyan, 2019-10-02:
> I just found out this error occurs when a friend is charging his iphone on my mac, immediately i disconnect it it starts working, strange right.

yes you found right. this is weird thing.

FunkyWhiteDude, 2019-12-25:
okay, so i am on a MacBook Pro 2017, so i am using an USB C to USB B connector, so i think this is causing my error -21, because i may make my macbook pro THINK it is charging another device, but it isn't. how could i solve this? is my case hopeless?

lmamane, 2022-09-23:
For people reading this, I had a "brown paper bag" moment. This error message also appears when the iDevice shows the popup "Trust this computer?" and one hasn't answered yet on the iDevice... (Or one has answered no, I suppose.)

JoshuaMacklin, 2025-05-01:
For anyone else that has run into this `idevicepair unpair && idevicepair pair` has worked for me