libimobiledevice/libimobiledevice #726: [BUG] idevicebackup2 restore fails on iPhone XS second time

kamstrup, 2018-10-22:
If I do a backup restore cycle on a new iPhoneXS it works as expected (with usbmuxd patches hinted to from https://github.com/libimobiledevice/libimobiledevice/issues/711). If I try to restore the same backup a second time it fails with:

```
...
Sending '00008020-001D099A1108002E/4a/4ab460371bb15855aa6dec30242079198c6573cf' (1.6 KB)
Sending '00008020-001D099A1108002E/55/550a09c4f8c4d89df203ab615ffc979d3c56f613' (780 Bytes)
Sending '00008020-001D099A1108002E/25/25cc406fbdae3703d891c2fa57d9d39930e0956e' (42 Bytes)
Sending '00008020-001D099A1108002E/77/77e6b5a266733afee2b161f9ec40f2164639ab21' (301 Bytes)
[==================================================] 100% Finished
ErrorCode 102: Error moving incomplete restore directory into place at path "/var/.backup.i" (MBErrorDomain/102). Underlying error: “.backup.i” couldn’t be moved to “var” because an item with the same name already exists. at path "/var/.backup.i" (NSCocoaErrorDomain/516). Underlying error: The operation couldn’t be completed. File exists (NSPOSIXErrorDomain/17)..
Restore Failed (Error Code 102).
```

If I do a factory reset of the phone I can again do a single backup/restore cycle before getting the error. My guess is that the act of restoring leaks a file in `/var/.backup.i` that should have been cleaned up.

Blefish, 2018-10-22:
I always restore with --reboot, which trigger device restart. Does this help?

kamstrup, 2018-10-22:
Thanks for the tip @Blefish -- it "fixes" it somewhat. After a reboot I can do a successful restore. Subsequent restores still reproduce the error, until I reboot.

kamstrup, 2018-10-22:
Reproduced this on an ipod with ios 11 as well, so it's not X/XS related

codebar33, 2018-10-22:
Yes, I confirm, if you do not pass the option --reboot, the temporary restore folder `/var/.backup.i` will remain on the device until your reboot it and therefore will prevent any other restore.

mexmer, 2018-10-22:
files in device are created by device process, not by libimobile, so you can say, it's issue inside iOS.
or maybe it's just failsafe protection, to force user to always restart after restore.

kamstrup, 2018-10-23:
Yeah, I've been thinking about this too... Maybe it's a bit much to expect that an OS can do a restore without rebooting (I know in theory it should be fine, but theory != reality in CS very often ;-P)

nikias, 2019-01-26:
See commit d200973897e281ba35d6c9c433e1355a49056da8.

closed 2019-01-26 by nikias