libimobiledevice/libimobiledevice #727: Package requirements (libusbmuxd >= 1.1.0) were not met during Flutter installation

nailgilaziev, 2018-10-23:
When installing flutter tool warning in terminal occurred:
```
==> ./autogen.sh
Last 15 lines from /Users/n/Library/Logs/Homebrew/libimobiledevice/01.autogen.sh:
checking dynamic linker characteristics... darwin18.0.0 dyld
checking how to hardcode library paths into programs... immediate
checking for pkg-config... /usr/local/opt/pkg-config/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libusbmuxd >= 1.1.0... no
configure: error: Package requirements (libusbmuxd >= 1.1.0) were not met:

Requested 'libusbmuxd >= 1.1.0' but version of libusbmuxd is 1.0.10

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libusbmuxd_CFLAGS
and libusbmuxd_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

READ THIS: https://docs.brew.sh/Troubleshooting
```

Full log here:
[libimobiledevice.log](https://github.com/libimobiledevice/libimobiledevice/files/2506066/libimobiledevice.log)

----
Flutter installation next: 
https://flutter.io/sdk-archive/#macos
Dev version 0.10.1

----
macOS Mojave 10.14,  MacBook pro 15 2018

nailgilaziev, 2018-10-23:
And seems this error cause a problem.
```
pro:local n$ flutter doctor
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel dev, v0.10.2-pre.20, on Mac OS X 10.14 18A391, locale en-RU)
[✓] Android toolchain - develop for Android devices (Android SDK 28.0.3)
[!] iOS toolchain - develop for iOS devices (Xcode 10.0)
    ✗ Verify that all connected devices have been paired with this computer in Xcode.
      If all devices have been paired, libimobiledevice and ideviceinstaller may require updating.
      To update, run:
        brew uninstall --ignore-dependencies libimobiledevice
        brew install --HEAD libimobiledevice
        brew install ideviceinstaller
[✓] Android Studio (version 3.2)
[!] IntelliJ IDEA Ultimate Edition (version 2018.2.4)
    ✗ Flutter plugin not installed; this adds Flutter specific functionality.
    ✗ Dart plugin not installed; this adds Dart specific functionality.
[✓] Connected device (1 available)

```

nailgilaziev, 2018-10-23:
And update is not help:
```
pro:~ n$ brew uninstall --ignore-dependencies libimobiledevice
Uninstalling /usr/local/Cellar/libimobiledevice/1.2.0_3... (66 files, 1MB)
pro:~ n$ brew install --HEAD libimobiledevice
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> Updated Formulae
prips                                                                              when

==> Cloning https://git.libimobiledevice.org/libimobiledevice.git
Updating /Users/n/Library/Caches/Homebrew/libimobiledevice--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at b34e343 tools: Remove length check on device UDID arguments to support newer devices
==> ./autogen.sh
Last 15 lines from /Users/n/Library/Logs/Homebrew/libimobiledevice/01.autogen.sh:
checking dynamic linker characteristics... darwin18.0.0 dyld
checking how to hardcode library paths into programs... immediate
checking for pkg-config... /usr/local/opt/pkg-config/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libusbmuxd >= 1.1.0... no
configure: error: Package requirements (libusbmuxd >= 1.1.0) were not met:

Requested 'libusbmuxd >= 1.1.0' but version of libusbmuxd is 1.0.10

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libusbmuxd_CFLAGS
and libusbmuxd_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

READ THIS: https://docs.brew.sh/Troubleshooting

pro:~ n$ brew install ideviceinstaller
Warning: ideviceinstaller 1.1.0_4 is already installed and up-to-date
To reinstall 1.1.0_4, run `brew reinstall ideviceinstaller`
pro:~ n$ flutter doctor
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel dev, v0.10.2-pre.20, on Mac OS X 10.14 18A391, locale en-RU)
[✓] Android toolchain - develop for Android devices (Android SDK 28.0.3)
[!] iOS toolchain - develop for iOS devices (Xcode 10.0)
    ✗ libimobiledevice and ideviceinstaller are not installed. To install, run:
        brew install --HEAD libimobiledevice
        brew install ideviceinstaller
[✓] Android Studio (version 3.2)
[!] IntelliJ IDEA Ultimate Edition (version 2018.2.4)
    ✗ Flutter plugin not installed; this adds Flutter specific functionality.
    ✗ Dart plugin not installed; this adds Dart specific functionality.
[!] Connected device
    ! No devices available

! Doctor found issues in 3 categories.
```

nailgilaziev, 2018-10-23:
these instructions helped me
https://github.com/flutter/flutter/issues/22595

kaushicramesh, 2018-11-13:
flutter/flutter#22595 worked for me as well

Enderllz, 2019-02-28:
flutter/flutter#22595 works really good

patelkeval78, 2019-10-12:
A recent change to libimobiledevice bumped the constraint on libusbmuxd to >= version 1.1.0. The current usbmuxd homebrew package is version 1.0.10.

As a result, homebrew --HEAD installs of libimobiledevice no longer build without a --HEAD install of usbmuxd.

Until the usbmuxd homebrew formula is updated, a workaround is to install it at HEAD:

brew update
brew uninstall --ignore-dependencies libimobiledevice
brew uninstall --ignore-dependencies usbmuxd
brew install --HEAD usbmuxd
brew install --HEAD libimobiledevice

See More at [Here](https://icetutor.com/question/libusbmuxd-version-error-during-flutter-install/)

Krunal79-flutter, 2019-10-15:
Already Solved [Here](https://icetutor.com/question/libusbmuxd-version-error-during-flutter-install/)