libimobiledevice/libimobiledevice #733: Can libimobiledevice support unified logging

rcmpayne, 2018-11-08:
Apple moved to unified logging from ASL back in 2010 and we are now seeing logs of developers disable the ASL logging with in the app making idevicesyslog kind of useless. What options do we have here?

https://www.bignerdranch.com/blog/migrating-to-unified-logging-swift-edition/
https://developer.apple.com/documentation/os/logging?language=objc

jumois, 2019-01-04:
Apple documentation suggests the **log** command line tool for viewing log messages, but it seems to be very limited compared to the GUI Console.app that can be used to view logs from connected iOS devices.

There seems to be no cli tools available for unified logging system?

rcmpayne, 2019-01-10:
The only option we have is grab a sysdiag from the device ( vol +, vol - and power). Once it's generated we can use current libimobiledevice tools to pull it. From here windows users need to ssh it to a Mac and run the ssh log command to decrypt the log and pipe to a file. 

This is the only workaround I've found thus far

christopherfujino, 2019-10-31:
So https://github.com/libimobiledevice/libimobiledevice/issues/861 was really this issue. Would it be possible to support unified logging @nikias?