libimobiledevice/libimobiledevice #735: has  "atc" service been Implemented ? can use libimobiledevice  sync music?

yinyajiang, 2018-11-19:
I only find mobilesync_* interface,but do't find interface realated "atc"(AirTrafficControl) service. I want to use libimoobiledevice for sync musit , not  itunes.

mexmer, 2018-11-19:
media synchronization is not supported in libimobile.

wolfteeth, 2022-06-02:
If we're waiting around for someone to implement this, I'm wondering, is there any documentation for what's needed to communicate with the ATC service? I imagine Apple keeps their details secret (and I didn't find anything with a superficial search in their developer docs), but surely someone has looked into it. It would be great even to have some guidance for figuring out the protocol. I suppose reverse engineering with something like Wireshark is part of it, but beyond that...