libimobiledevice/libimobiledevice #737: Stuck when trying to run idevicedebug on iOS 12

rermolov, 2018-11-26:
**What I do:**
Trying to run an installed app (iPhone 5s, iOS 12.1) with idevicedebug (compiled from the latest HEAD).

`idevicedebug run my_bundle_id`

**What I expect:**
Printing logs from application into console

**What I receive:**
idevicedebug is stuck and never returns.
If I run idevicedebug with debug, it becomes clear, that the problem is occuring when it receives command_prefix:

```
debugserver.c:386 debugserver_client_receive_response(): attempting to receive ACK +
debugserver.c:388 debugserver_client_receive_response(): received char: +
debugserver.c:398 debugserver_client_receive_response(): should_receive: 1, skip_prefix: 0
debugserver.c:401 debugserver_client_receive_response(): attempting to receive prefix
service.c:166 service_receive_with_timeout(): could not read data
debugserver.c:142 debugserver_client_receive_with_timeout(): Could not read data, error 0
debugserver.c:403 debugserver_client_receive_response(): received command_prefix: 
debugserver.c:414 debugserver_client_receive_response(): buffer: (null), should_receive: 0, skip_prefix: 0
debugserver.c:460 debugserver_client_receive_response(): response: (null)
debugserver.c:514 debugserver_client_send_command(): response result: 0
debugserver.c:520 debugserver_client_send_command(): received response: (null) 
```

Smmlove, 2019-03-08:
hi , excuse me, do you have solved the problem? I am meetting  the same problem, and I just want to start an app, not the log,  but "idevicedebug run my_bundle_id" stunck my other action

raspu, 2019-04-02:
I wasn't able to print the console log, but I was able to make the process end after the app is terminated, for some reason if you pass an environment variable that prints something it works: `idevicedebug -e DYLD_PRINT_STATISTICS=1 run com.comscore.NativeAnalyticsDriver`.