libimobiledevice/libimobiledevice #741: idevicesyslog not working with wifi device

softlion, 2018-12-12:
On macos mojave, lib installed with ```brew install --HEAD libimobiledevice ```, idevice_id correctly detects the device connected by wifi ('xcode devices' window also shows it as connected), but idevicesyslog is not finding the device. Any clue ?

```
> idevice_id --list
3e62372b2005xxxxxxxxxxx361b0f4c67764ff

> idevicesyslog    
Device with udid 3e62372b2005xxxxxxxxxxx361b0f4c67764ffnot found!?
Could not start logger for udid 3e62372b2005xxxxxxxxxxx361b0f4c67764ff
```

Any clue ?

nikias, 2018-12-12:
This is because usbmuxd lists the device but libimobildevice doesn't understand yet that it is a wifi device. I am working on proper support for it. Just be a little more patient.

vlar, 2018-12-15:
@nikias That would be amazing. 
I just installed it for this sole purpose, to get info over wifi. 

I saw on the website the status WIP and thought the project wasn't updated, so your comment is really cheering up.

softlion, 2019-04-08:
Hey :) Someone made it ?

trudog, 2019-06-20:
@nikias - any idea when this might be available?  thank you!

trinhpham, 2019-08-30:
It would remove a lot of wires on my table when I have several devices to be automated with my Macbook 13'

SCENEE, 2019-12-18:
Hi all! Now I've confirmed `idevicesyslog` is working with a device enabled "connect to network" on Xcode. I guess [this commit](https://github.com/libimobiledevice/libimobiledevice/commit/7dbc17f2c763a7b18bc1a867f39918c26bb5e9e5) fixed this issue. Thanks @nikias!