libimobiledevice/libimobiledevice #746: Is it possible to read unactivated eSIM (no PIN code written) Carrier info from eSIM iOS device?

taanders, 2019-01-08:
I tried to read device info with ideviceinfo. When there is set eSIM (Mobile data plan) to the device but eSIM PIN code is not written at boot then idevicerestore just passes line like there is no eSIM set to the device:
CarrierBundleInfoArray[0]:

If I set eSIM PIN code then device prints Carrier info.

Is it so that ideviceinfo just prints PLIST data from device and there is no such data about Carrier info available when eSIM PIN code is not set?

nemecence90, 2019-05-31:
Hey @taanders, did you find anything on this? I am trying to access EID for ESIM slots and CSN for iPads. Did anything come across for you that could be uselful for me?

taanders, 2019-06-03:
I have not found anything yet related this issue.

nemecence90, 2019-06-14:
Hey @nikias, can you help me out here buddy?

Dantee296, 2021-01-04:
@taanders  any update on this ?
also is it only with unactivated esim?
have u tried to get info via activated esim