libimobiledevice/libimobiledevice #748: App not visible in SpringBoard after install on iOS 12.1.x+ using ideviceinstaller

cmublitz, 2019-01-29:
Using:
libimobiledevice Master as of December 18, 2018 (there did not appear to be relevant commits after that date)
iOS 12.1.1
MacOS 10.13.6
iTunes 12.8.0

When I install an app via ideviceinstaller, the installation seems like it fully succeeds, but the app icon is not visible in SpringBoard.  I can start the app via the search function, it is visible in iTunes, and it is visible after restarting the iPad.  If I install the same app using iTunes, it is immediately visible like normal.  The app also installs fine on iOS 11.4+ devices using ideviceinstaller.

Blefish, 2019-01-29:
It seems like a bug with iOS, I was able to notice this. My workaround is to use springboardservices and explicitly set app icon location, this fixes the problem.

cmublitz, 2019-01-29:
Interesting, I don't know that I would have thought to do that.  Let me see if I understand what you are suggesting.  Something like:

lockdownd_start_service(client, "com.apple.springboardservices", &sbservice);
sbservices_client_new(device, sbservice, &sbc);
sbservices_get_icon_state(sbc, &iconstate, "2");
check the iconstate list and see that the new app is not present
**fabricate a new entry and add it to the iconstate list**
sbservices_set_icon_state(sbc, newiconstate);

Something like that?

nikias, 2019-01-29:
Totally an iOS bug. I wonder if this changes when I add the installation animation (the clockwise progress thingy)...

cmublitz, 2019-01-29:
I was able to verify the workaround mentioned by Blefish resulted in the icon being visible on an iPad page.

mexmer, 2019-01-29:
keep in mind, that apple removed application installation from itunes. technically you can still drag & drop app package to itunes, but since they removed stored and application tab from itunes itself. it's quite possible they dropping support for "outside" installation, except for developers trough xcode for testing.