libimobiledevice/libimobiledevice #750: Is there any way to check battery genuine genuineness without physically opening iPhone devices?

taanders, 2019-01-30:
Is there any way to check battery genuine genuineness without physically opening iPhone devices? Apple support pages show this thread:
https://support.apple.com/en-us/HT208387

There seems to be some kind of on iOS checks like:
Battery health unknown
and
Important Battery Message

Johnny314, 2019-05-29:
@taanders  you can achieve this like 3utools is doing right now

taanders, 2019-06-03:
Hi @Johnny314 do you know more details about reading this ex factory battery serial number and current battery serial number. Where those can be read and how? Some plist somewhere?

Johnny314, 2019-06-05:
not sure from where they ready ex factory battery serial number.
@nikias  please advice where to pick the ex factory battery serial value

bugHappy, 2019-06-21:
call fun diagnostics_relay_query_ioregistry_entry(diagnostics_client, "AppleARMPMUCharger", NULL, &node) can get battery serial value;  Battery health =NominalChargeCapacity/DesignCapacity

truonggiang0710, 2020-01-15:
Hi @taanders @nikias ,
Please advice how to get the ex factory battery serial value !
Thanks.

nikias, 2020-04-14:
@truonggiang0710 I would love to know myself, but I don't know this unfortunately.

Dantee296, 2020-04-24:
@taanders  u are looking for "SysCfg"

truonggiang0710, 2021-06-24:
@nikias I think we get SysCfg data to get the battery serial number.
On the iPhone Xs and above, the SysCfg data doesn't contain the ex-factory value of the battery serial number.
Do you know another way to check battery genuineness?

msk521, 2021-12-15:
@truonggiang0710 Can you tell me how you parse syscfg data？

ankit125664, 2023-05-31:
Have you achieved or not?

YakirSr, 2024-03-14:
please share if there is any clue about getting genuine parts data from ios mobile device.