libimobiledevice/libimobiledevice #751: List paired devices

didix21, 2019-01-30:
I was trying to get a list of the devices paired on my mac with the method `userpref_get_paired_udids()` but an empty list is returned when I have called it. I have used `lockdown_pair()` for pairing my device and it has succeed. After looking the method in depth I could see a comment saying that `userpref_get_paired_udids()` is deprecated. Thus, is there another way to list the udids ?

mexmer, 2019-01-30:
main issue with paired devices is, that since ios 9 or so, pairing records are only temporary, also pairing records don't work, unless you unlock device. so they are unrealiable.
you can list folder with pairing records in fact, but even then, it doesn't mean, all udids you find there will connect, without need of trust confirmation, or redoing pairing.

nikias, 2019-02-08:
Yes this capability is kind of pointless nowadays. It actually doesn't work anymore e.g. on macOS because the access permission to `/var/db/lockdown` won't allow it for a normal user to list the files.

didix21, 2019-02-19:
Ok thanks, I close the issue.

closed 2019-02-19 by didix21