libimobiledevice/libimobiledevice #752: Idevicepair validate is working only for older os versions but not for new os versions

jxm0418, 2019-02-05:
I tried to do "idevicepair validate" with new phone(iphone XR , OS -12.1.3) and it is throwing an error. But when I tried to do the same with old phone(iPhone 6s, OS - 10.2) it is working. 

I have provide the logs for your reference.  

M452:~ jxm0418$ idevice_id -l
9f140890c67da761e14a2f15e15c86e354fb3d92
00008020-00150D111AF8002E
M452:~ jxm0418$ idevicepair -u 9f140890c67da761e14a2f15e15c86e354fb3d92 validate
SUCCESS: Validated pairing with device 9f140890c67da761e14a2f15e15c86e354fb3d92
M452:~ jxm0418$ idevicepair -u 00008020-00150D111AF8002E validate
ERROR: Device 00008020-00150D111AF8002E returned unhandled error code -8

nikias, 2019-02-05:
Are you sure you are using the latest code?

jxm0418, 2019-02-05:
Yes I am using the latest code

nikias, 2019-02-06:
Can you run ideviceinfo? Also can you configure with --enable-debug-code and run `idevicepair -d validate` and/or `ideviceinfo -d` ?

jxm0418, 2019-02-07:
Last login: Thu Feb  7 09:36:03 on ttys000
M452:libimobiledevice jxm0418$ ideviceinfo
ActivationState: Activated
ActivationStateAcknowledged: true
BasebandActivationTicketVersion: V2
BasebandCertId: 165673526
BasebandChipID: 94
BasebandKeyHashInformation: 
 AKeyStatus: 64
 SKeyStatus: 2
BasebandMasterKeyHash: 1B41607650EBF11C6B39F41CB267DC64C121A9BCF44DBA5D28F55ACC86361BBA366554CD57B4C466055803E1EF81C870
BasebandRegionSKU: AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
BasebandSerialNumber: QcMtXuc2kikAAAAA
BasebandStatus: BBInfoAvailable
BasebandVersion: 1.03.08
BluetoothAddress: e4:b2:fb:21:5b:94
BoardId: 12
BrickState: false
BuildVersion: 16D40
CPUArchitecture: arm64e
CarrierBundleInfoArray[1]: 
 0: 
  CFBundleIdentifier: com.apple.Verizon_LTE_US
  CFBundleVersion: 35.0.2
  GID1: bae0000000000000
  GID2: ffffffffffffffff
  IntegratedCircuitCardIdentity: 89148000004521204639
  InternationalMobileSubscriberIdentity: 311480451650869
  MCC: 311
  MNC: 480
  MobileEquipmentIdentifier: 35734409230880
  SIMGID1: uuAAAAAAAAA=
  SIMGID2: //////////8=
  Slot: kOne
  kCTPostponementInfoAvailable: SIMCarrierInfo
CertID: 165673526
ChipID: 32800
ChipSerialNo: QcMtXuc2kikAAAAA
DeviceClass: iPhone
DeviceColor: 1
DeviceName: Jagadeesh Marella’s iPhone
DieID: 5925341628989486
EthernetAddress: e4:b2:fb:21:92:6e
FirmwareVersion: iBoot-4513.230.10
FusingStatus: 3
GID1: bae0000000000000
GID2: ffffffffffffffff
HardwareModel: N841AP
HardwarePlatform: t8020
HasSiDP: true
HostAttached: true
IntegratedCircuitCardIdentity: 89148000004521204639
InternationalMobileEquipmentIdentity: 357344092308809
InternationalMobileEquipmentIdentity2: 357344092386227
InternationalMobileSubscriberIdentity: 311480451650869
InternationalMobileSubscriberIdentityOverride: false
MLBSerialNumber: C7H84170TZCL1P145
MobileEquipmentIdentifier: 35734409230880
MobileSubscriberCountryCode: 
MobileSubscriberNetworkCode: 480
ModelNumber: MT312
NonVolatileRAM: 
 auto-boot: dHJ1ZQ==
 backlight-level: MTQxNQ==
 boot-args: 
 bootdelay: MA==
 com.apple.System.tz0-size: MHg4RDAwMDAw
 oblit-begins: T2JsaXRUeXBlOiBPYmxpdGVyYXRlRGF0YVBhcnRpdGlvbi4gUmVhc29uOiB1bmtub3du
 obliteration: aGFuZGxlX21lc3NhZ2U6IE9ibGl0ZXJhdGlvbiBDb21wbGV0ZQo=
PRIVersion_Major: 0
[... 163 lines cut ...]
    V2lmV1JLSGpoL1dCbG1jMkJ5eFRHdTBCbytZUGNvU2ZLcS9kTG5td3RjWjFx
    NWoKM0ZsTEdlOG9pYlN2eUZTTXBpWjJkR1RHdnNPMDF3TGpCYmlJK00xbjEz
    bDlGUmlIbHhnL05PclAxUjBqcytaRQpqWjVvTDJZa0djcVZDY0N6WHF5enQ1
    TXpyRTBmTGlvSFQvajZHNGNCcWFGVnNzZ2xHS0RkcmxJeDFydnQ4SWlaCm1W
    a21MTXZxV2Rmek5ZZVB2RWxrTHc1YmlvTUhhLzFHWUVKbEYxUnRTK2RHZjVj
    ZCtrczZrTDlWdnRab3dtcHYKSTJ3ZTYwdDFZUDQzaUUrdW03ME5YRXl5aU5h
    RHUxQkFWanVyRUlFM1NBcXN5WFpCUUx0c1RMeVdzbmYrUi8vbgp0bkY0Mmc5
    aC9GQjNDUjVzdmtzUVA2TT0KLS0tLS1FTkQgQ0VSVElGSUNBVEUtLS0tLQo=
        </data>
        <key>SystemBUID</key>
        <string>6F511C79-3A18-4EF0-B7DF-ACA8EEACF9C3</string>
        <key>WiFiMACAddress</key>
        <string>e4:b2:fb:35:b9:ff</string>
    </dict>
    <key>Request</key>
    <string>ValidatePair</string>
    <key>ProtocolVersion</key>
    <string>2</string>
</dict>
</plist>
15:50:05 idevice.c:398 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -35 (No such file or directory)
15:50:05 service.c:163 service_receive_with_timeout(): could not read data
15:50:05 property_list_service.c:193 internal_plist_receive_timeout(): initial read=0
15:50:05 property_list_service.c:195 internal_plist_receive_timeout(): initial read failed!
ERROR: Device 00008020-00150D111AF8002E returned unhandled error code -8

nikias, 2019-02-08:
Oh!!! I know. Apple removed the ValidatePair command from lockdownd. See my commit message from  August 13, 2017:
```
commit 5a85432719fb3d18027d528f87d2a44b76fd3e12
Author: Nikias Bassen <nikias@gmx.li>
Date:   Sun Aug 13 00:12:16 2017 +0200

    lockdown: Don't explicitly validate pairing unless we're dealing with an older device

    On newer iOS version, ValidatePair is not mandatory to gain trusted host
    status. Starting with iOS 11, the ValidatePair request has been removed from
    lockdownd and will throw an error. This commit adds a version check so that
    ValidatePair is only called on devices prior iOS 7.
```
Please understand that the tool idevicepair technically is not required anymore since usbmuxd is handling the pairing. Nevertheless I will supply a fix for idevicepair.

nikias, 2019-02-08:
Fixed with 8ef9e51d8514a441dc9b7d035954691d880a543e.

closed 2019-02-08 by nikias

jxm0418, 2019-02-08:
hey, This fix is not completely working.

M452:~ jxm0418$ idevicepair validate
ERROR: Device 8a28726f4e641c95e5df4f9c38fb87fe256b8c47 returned unhandled error code -2
M452:~ jxm0418$ idevicepair pair
SUCCESS: Paired with device 8a28726f4e641c95e5df4f9c38fb87fe256b8c47
M452:~ jxm0418$ idevicepair validate
SUCCESS: Validated pairing with device 8a28726f4e641c95e5df4f9c38fb87fe256b8c47
M452:~ jxm0418$ idevicepair unpair
SUCCESS: Unpaired with device 8a28726f4e641c95e5df4f9c38fb87fe256b8c47
M452:~ jxm0418$ idevicepair validate
ERROR: Device 8a28726f4e641c95e5df4f9c38fb87fe256b8c47 returned unhandled error code -2

nikias, 2019-02-08:
@jxm0418 I noticed your reply that you deleted. Did it work now?

jxm0418, 2019-02-09:
when the device is trusted it is good.
M452:~ jxm0418$ idevicepair validate
SUCCESS: Validated pairing with device 9f140890c67da761e14a2f15e15c86e354fb3d92

But when the device is not trusted. It is giving unhandled error -2 . I think it should throw error like "not paired" or something meaningful message.

M452:~ jxm0418$ idevicepair validate
ERROR: Device 9f140890c67da761e14a2f15e15c86e354fb3d92 returned unhandled error code -2

nikias, 2019-02-09:
See 797a9e7e79d86427f40dbb82d271125b8c0b2485.