libimobiledevice/libimobiledevice #756: Disconnect/Forget devices

imoby, 2019-02-14:
Is it possible to forget a specific device that is connected? For example, running 
```idevice_id -l``` will return a list of connected devices. If I wanted to disconnect them so that they are no longer recognized (user has to unplug/plug to recognize again), how would I achieve this.

nikias, 2019-02-14:
Use `idevicepair -u UDID unpair`.

imoby, 2019-02-14:
@nikias 

Thank you so much! I wasnt aware that there was a idevicepair too. I was able to successfully unpair the device but I think I have a followup question regarding what idevice_id provides vs what idevicepair does/doesnt do.

My understanding is that the idevice_id lists out all the devices that are connected to the machine. idevicepair however lists all the 'paired' devices. However, its possible to not need to pair before doing a backup/restore correct?

Also, running this command ```idevicepair --list``` never returns me anything whether I pair or unpair any devices. Am I missing something?

nikias, 2019-02-14:
idevicepair list doesn't really work anymore. It's a permission problem; it used to list the pairing record files to get an idea what devices have been paired on the current machine, but e.g. on macOS Apple changed the permissions so this doesn't work anymore (unless you are root). But there's even more than that: pairing records expire now, so the output won't reflect truly valid pairings. In the end the devices decide which records are valid or not ;)

imoby, 2019-02-15:
Thank you so much! @nikias you've been very helpful. One last question I have, is it possible to ask the 'trust' question again after the user has declined it?

closed 2020-10-27 by imoby