libimobiledevice/libimobiledevice #757: Info on Wifi Sync?

vlar, 2019-02-16:
Hello, 

I'm seeing a lot of updates recently. I was wondering whether it is possible to sync WIFI iOS device?

fosple, 2019-02-23:
I would also be very interested in this feature. Any updates? :)

VaporX88, 2019-02-24:
As far as i can read about it nikias is working on it. If i am right Usbmuxd support this allready but the tools has to be changed to a new API.

kazzmir, 2019-03-08:
This is a one line change in src/idevice.c:
```
// int res = usbmuxd_get_device_by_udid(udid, &muxdev);
+        int res = usbmuxd_get_device(udid, &muxdev, DEVICE_LOOKUP_USBMUX | DEVICE_LOOKUP_NETWORK);
``` or the API 's could be changed such that the client specifies whether to its ok to look up network devices.

fosple, 2019-03-10:
Thanks @kazzmir, I see there is already a PR open for this:
https://github.com/libimobiledevice/libimobiledevice/pull/728