libimobiledevice/libimobiledevice #759: Detect if device is jailbroken

marchuk-nikolay, 2019-02-18:
Hello!
Could you say, please, is there any way to determine if device is jailbroken using libimobiledevice?

I can do it for device with iOS 7.1.2, trying to start afc2 service. But for device with iOS 11.4.1 this approach doesn't work.

nikias, 2019-02-18:
Nothing I am aware of. Sorry.

Kanong017, 2019-02-19:
You can check if there is a Cydia icon on the device.
Using sbservice service

marchuk-nikolay, 2019-02-19:
@nikias , @Kanong017 , thanks for answers. I've found solution similar to @Kanong017 , but a little bit another. We can get list of system apps (using ideviceinstaller tool) and if it contains Cydia, we can suggest that device is jailbroken.

nikias, 2019-02-19:
That works, but technically it doesn't give you a 100% correct result because it could also be jailbroken without having Cydia installed. Just keep that in mind ;)