libimobiledevice/libimobiledevice #761: configure: error:    Could not link test program to Python.

xguntis, 2019-02-25:
Hi,
I'm on Ubuntu 18.04. Have installed the Python. Nothing fancy. Standard installation. And i get the following error at the end of executing ./autogen.sh

```
checking for a Python interpreter with version >= 2.3... python
checking for python... /usr/bin/python
checking for python version... 2.7
checking for python platform... linux2
checking for python script directory... ${prefix}/lib/python2.7/dist-packages
checking for python extension module directory... ${exec_prefix}/lib/python2.7/dist-packages
checking for cython... no
configure: WARNING: cannot find 'cython' program. You should look at http://www.cython.org or install your distribution specific cython package.
checking for python2.7... (cached) /usr/bin/python
checking for a version of Python >= '2.1.0'... yes
checking for the distutils Python package... yes
checking for Python include path... ./configure: line 18316: type: /usr/bin/python-config: not found
-I/usr/include/python2.7
checking for Python library path... ./configure: line 18336: type: /usr/bin/python-config: not found
-L/usr/lib/python2.7 -lpython2.7
checking for Python site-packages path... /usr/lib/python2.7/dist-packages
checking python extra libraries... ('-L/usr/lib -lz', '-lpthread -ldl  -lutil')
checking python extra linking flags... -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions
checking consistency of all components of python development environment... no
configure: error: 
  Could not link test program to Python. Maybe the main Python library has been
  installed in some non-standard library path. If so, pass it to configure,
  via the LDFLAGS environment variable.
  Example: ./configure LDFLAGS="-L/usr/non-standard-path/python/lib"
  ============================================================================
   ERROR!
   You probably have to install the development version of the Python package
   for your distribution.  The exact name of this package varies among them.
  ============================================================================
```

mexmer, 2019-02-27:
you have only python binaries installed, not python development package, you need to install python development package

dieash, 2019-11-11:
Hello, guys!

On the last week I got the same problem as you describe here. After deep diving in 'configure'-script I learn the following. 

1. There is a last-guard-check which looks like
```gcc -o conftest -g -O2 -pthread -I/usr/include/python2.7  conftest.c  -lboost_unit_test_framework -L/usr/lib -lpython2.7 -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions -L/usr/lib -lz -lpthread -ldl  -lutil```

2. And result of this command:
```/usr/bin/x86_64-linux-gnu-ld: cannot find -lz```

3. The solution is very simple in my case:
```apt-get install zlib1g-dev```

May be it helps you or someone else with such problem.
Have a nice day!

tolgasel, 2020-03-23:
> Hello, guys!
> 
> On the last week I got the same problem as you describe here. After deep diving in 'configure'-script I learn the following.
> 
> 1. There is a last-guard-check which looks like
>    `gcc -o conftest -g -O2 -pthread -I/usr/include/python2.7  conftest.c  -lboost_unit_test_framework -L/usr/lib -lpython2.7 -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions -L/usr/lib -lz -lpthread -ldl  -lutil`
> 2. And result of this command:
>    `/usr/bin/x86_64-linux-gnu-ld: cannot find -lz`
> 3. The solution is very simple in my case:
>    `apt-get install zlib1g-dev`
> 
> May be it helps you or someone else with such problem.
> Have a nice day!
> 
> PS
> It was another project, not yours. But autotools.

Thanks! i had a problem with compiling apparmor and your comment (point 3) helped me to solve this python issue!

mortensassi, 2021-03-08:
For anyone still having issues with this; i have found out that i had installed python, python2 and python3. After uninstalling python2 `sudo apt-get purge python2.7-minimal` the problem disappeared. Please make sure nothing depends on python2 for you. Hope this works!

mh-cbon, 2022-04-03:
on fedora try `python3-devel`