libimobiledevice/libimobiledevice #762: install libimobiledevice error

lawDavid, 2019-02-26:
I use `brew install --HEAD libimobiledevice` to install libimobiledevice then generated some errors that I can't deal with, can someone help me?
```bash
iMac:~ davidLaw$ brew install --HEAD libimobiledevice
Updating Homebrew...
==> Cloning https://git.libimobiledevice.org/libimobiledevice.git
Updating /Users/davidLaw/Library/Caches/Homebrew/libimobiledevice--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 0584aa9 idevicepair: Fix udid being freed before printing error message containing udid
==> ./autogen.sh
==> ./configure --disable-silent-rules --prefix=/usr/local/Cellar/libimobiledevice/HEAD-0584aa9_3 --without-cython --enable
==> make install
Last 15 lines from /Users/davidLaw/Library/Logs/Homebrew/libimobiledevice/03.make:
        int res = pthread_create(thread, NULL, thread_func, data);
                                         ^
thread.c:52:23: error: use of undeclared identifier 'NULL'
        pthread_join(thread, NULL);
                             ^
thread.c:61:28: error: use of undeclared identifier 'NULL'
        pthread_mutex_init(mutex, NULL);
                                  ^
7 errors generated.
make[1]: *** [thread.lo] Error 1
make[1]: *** Waiting for unfinished jobs....
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.0.0_1/include -I/usr/local/Cellar/usbmuxd/HEAD-c75605d_1/include -I/usr/local/Cellar/libplist/2.0.0_1/include -I/usr/local/Cellar/openssl/1.0.2q/include -D_LARGE_FILES=1 -g -O2 -c socket.c -o socket.o >/dev/null 2>&1
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.0.0_1/include -I/usr/local/Cellar/usbmuxd/HEAD-c75605d_1/include -I/usr/local/Cellar/libplist/2.0.0_1/include -I/usr/local/Cellar/openssl/1.0.2q/include -D_LARGE_FILES=1 -g -O2 -c debug.c -o debug.o >/dev/null 2>&1
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I.. -I../include -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fsigned-char -fvisibility=hidden -I/usr/local/Cellar/libplist/2.0.0_1/include -I/usr/local/Cellar/usbmuxd/HEAD-c75605d_1/include -I/usr/local/Cellar/libplist/2.0.0_1/include -I/usr/local/Cellar/openssl/1.0.2q/include -D_LARGE_FILES=1 -g -O2 -c userpref.c -o userpref.o >/dev/null 2>&1
make: *** [install-recursive] Error 1
```

JCCP3, 2019-02-26:
same problem

ergunkocak, 2019-03-22:
Did anyone has a solution?