libimobiledevice/libimobiledevice #764: libimobiledevice on mac hangs causing jenkins jobs to time out.

mikelupo, 2019-03-19:
After a few days of the mac (Mojave) being used in iOS development as a Jenkins slave, the library becomes unusable.

For example;
idevice_id -l
ERROR: Unable to retrieve device list!  <-- this occurs after a timeout period.

There are 6 devices connected to this mac via a powered USB hub. 
If I cycle the power on the USB hub, libimobiledevice commands start working OK again.
Each device as used by the jenkins node will use libimobiledevice to reboot the device before testing begins. 

This problem began after updating to Mojave, and hence updating libimobiledevice since last year's libimobiledevice compilation wasn't working. It was working for years in this configuration before Mojave.

If there is a way to debug this, I'd be happy to in order to provide a more meaningful issue that could be understood and subsequently fixed.

All components were built and installed on the mac from scratch on 26 Feb 2019 from the latest HEAD on master.
libimobiledevice: 0584aa90c93ff6ce46927b8d67887cb987ab9545
libplist: bec850fe399639f3b8582a39386216970dea15ed
libusbmuxd: c75605d862cd1c312494f6c715246febc26b2e05
openssl: b34cf4eb616446a1ee7bd0db0a625edf25047342 (HEAD, tag: OpenSSL_1_0_2r)

OpenSSL 1.0.2r  26 Feb 2019
built on: reproducible build, date unspecified
platform: darwin64-x86_64-cc
options:  bn(64,64) rc4(ptr,int) des(idx,cisc,16,int) idea(int) blowfish(idx) 
compiler: cc -I. -I.. -I../include  -DOPENSSL_THREADS -D_REENTRANT -DDSO_DLFCN -DHAVE_DLFCN_H -arch x86_64 -O3 -DL_ENDIAN -Wall -DOPENSSL_IA32_SSE2 -DOPENSSL_BN_ASM_MONT -DOPENSSL_BN_ASM_MONT5 -DOPENSSL_BN_ASM_GF2m -DSHA1_ASM -DSHA256_ASM -DSHA512_ASM -DMD5_ASM -DAES_ASM -DVPAES_ASM -DBSAES_ASM -DWHIRLPOOL_ASM -DGHASH_ASM -DECP_NISTZ256_ASM

nikias, 2019-03-19:
It appears that you are using the default macOS usbmuxd, in that case if `idevice_id -l` is not returning anything this is definitely not an issue with libimobildevice since it just queries usbmuxd directly to get a list of devices. If the request takes longer and returns an error it seems that the (original) usbmuxd is not running at that time so you need to verify this first.

mikelupo, 2019-03-19:
Hi @nikias, Thanks for the quick reply. Is there any simple way to identify which usbmuxd is being utilized? I had compiled and installed the one (libusbmuxd) from this repo and assumed that it was the one that is being used.

nikias, 2019-03-19:
libusbmuxd is just the library that connects to whatever usbmuxd is available on the system. In fact it doesn't even matter which one is used, but the open source one could have a bug, while Apple's probably doesn't. You can check using this command, if it shows a the same path then this is Apple's:
```
$ ps axgu |grep usbmuxd
_usbmuxd           109   0,2  0,0  4377020   3980   ??  Ss    4Mär19  14:12.93 /System/Library/PrivateFrameworks/MobileDevice.framework/Versions/A/Resources/usbmuxd -launchd
```

mikelupo, 2019-03-19:
Another Mojave machine with the libimobiledevice waiting for the usbmuxd to return. My PS returns a that process running on a low number too. 115. Same path as you displayed so it's definitely Apples.
I'm going to update from 10.14 to 10.14.3 and see if that makes a difference. Probably an iTunes update wouldn't hurt either.

nikias, 2019-03-19:
Weird. Does iTunes show the devices?

mikelupo, 2019-05-21:
@nikias, no. iTunes also does not show devices.

nikias, 2019-05-22:
If iTunes doesn't show the devices this is certainly not a bug in libimobiledevice. Does iTunes (and idevice* tools) show the device if you plug one of them directly, without the hub? If so, the hub seem to be the cause of this. Run the System Profiler app and under Hardware select USB, then on the right side upper tree view check if you can see any iDevice.

mikelupo, 2019-07-01:
@nikias, this is also happening on machines with only two devices connected (no hub).
It's Apple. I agree, not libimobiledevice.

But it's good that others can reference this incase they encounter as well...
You can't even reboot the machine successfully via terminal. Hangs during shutdown part of the reboot. tried "sudo shutdown -r now" as well as "sudo reboot". Same outcome.
Since this happens every few days, I'll try adding a workaround in that I'll reboot each slave machine once every 24 hours. If it's not in that weird state, the machine will reboot successfully.

Thanks for listening.

closed 2019-07-01 by mikelupo