libimobiledevice/libimobiledevice #765: using idevicediagnostics to read battery information with IOS update 12.2

MatthewPerrison, 2019-03-27:
Hi, 

I am not sure at all if this is the right Place for this , but I hope it is .

I just updated 12.2 IOS to my Iphone XR and after that I am not able to read the battery's full charge capacity from that device anymore. Value is always 100 , and as I checked other devices as well, value is the same 100 in all the devices with 12.2 IOS update

in 12.1.4 , getting valid full charge capacity was still possible.

nikias, 2019-03-27:
Technically this is not the right place because this library and tools can only return what the device is providing. If the output changed then Apple is to blame here. Only question here is if Apple moved that information to a different location.

closed 2019-03-28 by MatthewPerrison

omkarzunjurke, 2019-03-29:
I also observed the same issue but we have different findings, I have updated iPhone Xs, iPhone X, iPhone 6s Plus to ios 12.2 but got this issue only on iPhone x and iPhone xs i.e we have still fetch the data from iPhone 6s plus with ios 12.2.

ericfri, 2019-04-04:
I've just started playing with this library, and in my search for the battery full charge info, I've got the same problem. I'm using an iPhone 5S with iOS 12.2 and when I hook it up to CoconutBattery, it reports that there is 93% of the design capacity remaining, and when I use 'idevicediagnostics diagnostics' it reports 100 for the Full Charge Capacity.

CoconutBattery shows:
Current Charge 1309 mAh
Full Charge Capacity 1460 mAh
Design Capacity 1560 mAh

The only number I've been able to get via this library is the Design Capacity (which is actually different, I get 1550 vs CoconutBattery's 1560).

Am I doing something wrong? Is this even possible to get? It must be, as CoconutBattery has it, but I'd like to put it into a script for my technicians to run on customers phones.

Matszs, 2019-04-06:
Having the same problem on an iPhone 7, iOS 12.2 is returning a value of 100. iOS12.1 is still working.

jackczhang, 2019-04-10:
Having the same issue on iOS 12.2 
:(

TheWeirdDev, 2020-06-30:
Same issue on iPadOS 13.2.3

bhakimi, 2020-11-12:
has this been addressed yet? i also have the same issue

mexmer, 2020-11-13:
it's not library problem, it's apple problem, they broke it in newer versions of iOS, or maybe they just changed location where value is stored, who knows.

bhakimi, 2020-11-14:
@mexmer not sure if im looking right, im trying to find out how much battery there is left, is there any other way to get this value like CurrentCapacity or maybe inside this`;

```
                <key>AbsoluteCapacity</key>
                <integer>1720</integer>
                <key>AdapterDetails</key>
                <dict>
                        <key>Current</key>
                        <integer>1500</integer>
                        <key>Description</key>
                        <string>usb charger</string>
                        <key>FamilyCode</key>
                        <integer>-536854523</integer>
                        <key>PMUConfiguration</key>
                        <integer>1490</integer>
                        <key>Voltage</key>
                        <integer>5000</integer>
                        <key>Watts</key>
                        <integer>7</integer>
                </dict>
```

mexmer, 2020-11-18:
not any i'm aware off.

puffhoister, 2022-04-24:
> I've just started playing with this library, and in my search for the battery full charge info, I've got the same problem. I'm using an iPhone 5S with iOS 12.2 and when I hook it up to CoconutBattery, it reports that there is 93% of the design capacity remaining, and when I use 'idevicediagnostics diagnostics' it reports 100 for the Full Charge Capacity.
> 
> CoconutBattery shows: Current Charge 1309 mAh Full Charge Capacity 1460 mAh Design Capacity 1560 mAh
> 
> The only number I've been able to get via this library is the Design Capacity (which is actually different, I get 1550 vs CoconutBattery's 1560).
> 
> Am I doing something wrong? Is this even possible to get? It must be, as CoconutBattery has it, but I'd like to put it into a script for my technicians to run on customers phones.
> 
> Edit. I just checked with an iPhone 6S on iOS 12.1 and it reports: Design Capacity: 1690 Full Charge Capacity: 1400

Did you found the solution?