libimobiledevice/libimobiledevice #769: brew install --HEAD libimobiledevice  Error: An exception occurred within a child process:

testerSunshine, 2019-04-11:
An error was reported while performing this step

```
==> Cloning https://git.libimobiledevice.org/libimobiledevice.git
Cloning into '/Users/huolala/Library/Caches/Homebrew/libimobiledevice--git'...
error: RPC failed; curl 56 LibreSSL SSL_read: SSL_ERROR_SYSCALL, errno 60
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
Error: An exception occurred within a child process:
  DownloadError: Failed to download resource "libimobiledevice"
Failure while executing; `git clone --branch master https://git.libimobiledevice.org/libimobiledevice.git /Users/huolala/Library/Caches/Homebrew/libimobiledevice--git` exited with 128. Here's the output:
Cloning into '/Users/huolala/Library/Caches/Homebrew/libimobiledevice--git'...
error: RPC failed; curl 56 LibreSSL SSL_read: SSL_ERROR_SYSCALL, errno 60
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
```
this url access  HTTP code is 403
```
https://git.libimobiledevice.org/libimobiledevice.git 
```

mexmer, 2019-04-11:
this is problem with github servers, not libimobile

testerSunshine, 2019-04-12:
thanks，I compiled it directly from the source code.
please Help me with #770  this problem.

closed 2019-04-12 by testerSunshine