libimobiledevice/libimobiledevice #772: Partial Backup Option

francescofact, 2019-04-22:
Hi,
Is it possible with this tool to do a partial backup of my iPhone? 
I already have my photos and other stuff on the iCloud library. I have a huge photo library, like 60gb, so I would like  to backup only Applications Data. Is it possible with this tool or also with a workaround?

Thank you

nikias, 2019-04-22:
Doesn't work since the device decides what to back up.

ibash, 2019-12-21:
@nikias I've been digging into this and it is kind of possible, though it would take roughly twice as long to run. The trick would be to run the backup in two passes.

1. First pass: run a backup, but instead of writing out data files only create the files (with 0 bytes). This would take minimal space on disk, but would take the full sync time.
2. After the first pass has run, inspect `Manifest.db` to determine which files to ignore and which to save.
3. Run the full backup again, but this time only write files desired from (2). Note that this still takes the same amount of time as a full backup, since the device will transmit everything.
4. Incremental backups after that will run at incremental speed! You can truncate files that are not desired.

In the end you end up with a directory structure that looks like a complete backup, but most of the files are empty.

Some notes:
1. The device does a filesystem scan of a backup, and if it notices a missing file, will do a complete backup. Which is why you need to have empty files.
2. Since the manifest is written last, and AFAICT that's needed for doing an incremental backup, a full sync needs to be done at _some_ point. If there was some way to create or spoof a manifest, it might be possible to get around this.
3. Since the hash is deterministic based off of domain / relative path, you could sync specific files (e.g. the call logs db) in a single pass.

mexmer, 2019-12-31:
@ibash you can write own app, that does this "2 step" backup. but ios itself doesn't offer such thing as partial backup and restore. also when you will be at it, you can already scrap files at first run, no need to do backup twice just to pull the list (manifest database), if you know which files or appcontainer you looking for, you can already filter files as they come. 

you can build your own tool for this on top of libimobiledevice library.

ibash, 2019-12-31:
@mexmer is there a way to pull the manifest without backing up all the files initially? I thought the manifest was only written at the end of the first backup.

nikias, 2019-12-31:
@ibash I think it sends it at the end. That's how the backup system works, and you can't request files selectively.

mexmer, 2019-12-31:
no, you can't pull manifest before doing backup. but during backup device sends name of file in phone + name it should be saved on your computer (hash), so you can filter files on reception by names.

in my software i log those, so it looks like this
```
receiving backup file "/.ba/mobile/Library/Preferences/com.apple.aggregated.plist", destination "Snapshot/61/61e1e4600e3997adbadda920ade1321e753b2b17"
```
so you can filter files on reception from phone already, without need of pulling out manifest. if you know which files you want, that is.

ibash, 2019-12-31:
Excellent, thank you!

Islam Sharabash
217-377-9657

Sent via Superhuman iOS ( https://sprh.mn/?vip=islam.sharabash@gmail.com )

sKerry-not-scarry, 2020-10-14:
Hi @ibash , we you able to reduce the sync time? I've done the same thing but my challenge now is to reduce the time.

ibash, 2020-10-14:
@sKerry-not-scarry nope, stopped working on this a while back. AFAIK you have to go through a full sync, but would be great if that could be avoided.

One thing to investigate would be iCloud backups, maybe there's something useful there.

kevinant2, 2022-09-05:
@mexmer 

Hi Mexmer, 

It's been a while but just would like to know how you modify the code so that it outputs 

receiving backup file "/.ba/mobile/Library/Preferences/com.apple.aggregated.plist", destination "Snapshot/61/61e1e4600e3997adbadda920ade1321e753b2b17"

is dname the (path name of the file) and fname is the hash name of the file in the mb2_handle_receive_files subroutine?

Thanks ,

Kevin

mexmer, 2022-09-05:
yes, put it into that function, i would put it here
https://github.com/libimobiledevice/libimobiledevice/blob/master/tools/idevicebackup2.c#L1054
something like this should be enough
```
printf("INFO: receiving %s, saving to %s", fname, dname);
```

kevinant2, 2022-09-06:
> yes, put it into that function, i would put it here https://github.com/libimobiledevice/libimobiledevice/blob/master/tools/idevicebackup2.c#L1054 something like this should be enough
> 
> ```
> printf("INFO: receiving %s, saving to %s", dname, fname);
> ```

Thanks Mexmer, will try this and hopefully it works