libimobiledevice/libimobiledevice #773: Could not start service com.apple.mobile.notification_proxy

UnitedMarsupials, 2019-04-26:
Trying to use the latest code -- on FreeBSD. I get:
```
root@aldan:~mi (1050) idevicepair unpair && idevicepair pair && idevicebackup2 backup MT-iPhone-backup
SUCCESS: Unpaired with device 6a85e740xxx88cee06
SUCCESS: Paired with device 6a85e740xxx88cee06
Backup directory is "MT-iPhone-backup"
ERROR: Could not start service com.apple.mobile.notification_proxy.
Segmentation fault (core dumped)
```
That's using libimobiledevice as of April 3rd this year.

nikias, 2019-04-27:
First, you don't need to unpair and pair manually, usbmuxd is handling that. But that's not the reason for the crash.

Can you run the command in gdb?
```gdb --args idevicebackup2 backup MT-iPhone-backup```
When it loaded, type ```r``` + ENTER, this will run the program.
When it prints Segmentation Fault / SIGSEGV, type ```bt``` + ENTER, then copy and paste the output here.

AlphaJack, 2021-08-01:
```bash
(gdb) r
Starting program: /usr/bin/idevicebackup2 backup MT-iPhone-backup
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Backup directory is "MT-iPhone-backup"
ERROR: Could not start service com.apple.mobile.notification_proxy.

Program received signal SIGSEGV, Segmentation fault.
0x0000555555557a35 in ?? ()
(gdb) bt
#0  0x0000555555557a35 in ?? ()
#1  0x00007ffff7d8db25 in __libc_start_main () from /usr/lib/libc.so.6
#2  0x000055555555b75e in ?? ()
```

nikias, 2021-08-01:
Can you pull the latest code and compile & test again? I pushed a few changes a few days ago specifically for idevicebackup2.

AlphaJack, 2021-08-02:
Tried with 24abbb9450c723617e10a6843978aa04a576523e, it works!

nikias, 2021-08-02:
Great. I also think I hit that exact segmentation fault, that's why I changed the code a bit.

closed 2021-08-02 by nikias

UnitedMarsupials, 2021-08-23:
The good news is, with today's master, this error no longer causes the tool to _crash_. The bad news is, I still cannot backup this phone -- an older 5S running iOS-12.5.4. It either gives this error, or the: `Could not start service com.apple.afc: Receive timeout`.

Even after a fresh reboot of the phone.

nikias, 2021-08-23:
Did you make sure to also update libusbmuxd?

UnitedMarsupials, 2021-08-23:
> Did you make sure to also update libusbmuxd?

Just tried upgrading libusbmuxd -- and usbmuxd itself too. No perceptible improvement. Tried against three different phones:
|Model|iOS version|Symptoms|
|------|-----|-----|
|iPhone-5C|10.3.3|This oldest one gets farthest -- _sometimes_. After looping all night, rebooting occasionally, I found it succeeding in downloading thousands of files (once), then hanging on: `Sending '6d71...38c6/Manifest.db' (9.9 MB)`. The little circled-arrows symbol on the phone is spinning, but nothing else is changing.
|iPhone-5S|12.5.4|`ERROR: Could not connect to lockdownd, error code -4` or `ERROR: Could not start service com.apple.mobile.notification_proxy: Unknown Error`
|iPhone-7|14.7.1|Either the `ERROR: Could not connect to lockdownd, error code -4` or `ERROR: Could not start service com.apple.afc: Receive timeout` or `Could not perform backup protocol version exchange, error code -1`
|iPad Pro|14.6|Not connecting -- not listed by `idevice_id`

With the possible exception for iPad, I don't think, these are _separate_ problems...