libimobiledevice/libimobiledevice #774: Airplane Mode turn on and off

Imtiazkarimik32, 2019-05-01:
Is it possible to turn airplane mode on and off of the device? If not, can you please give me any pointers on how to achieve this?

nikias, 2019-05-01:
You can't. And unless the device is jailbroken an app on the device won't have the permissions to achieve this.

Imtiazkarimik32, 2019-05-01:
I have jailborken the phone. Is there any command or method I can use?

Imtiazkarimik32, 2019-05-13:
Any suggestions will be highly appreciated.

Mgaskmg32org, 2019-06-06:
@Imtiazkarimik32 I got 2 but not sure if works at the moment: 

1.
If [[ $(ifconfig en0) != *inet* ]]

2.
If [[ $(ifconfig pdp_ip0) == *inet* ]]

Mgaskmg32org, 2019-06-06:
Or try this:

>/tmp/airplane mode-on ; rm -f 
/tmp/airplanemode-off

Btw connect those two in one line ^

2.
To turn off

>/tmp/airplanemode-off ; rm -f /tmp/airplanemode-on

^Connect Two lines to one line

Imtiazkarimik32, 2019-06-06:
@Mgaskmg32org Thank you very much for helping me.

Are these libimobiledevice commands? or I should add these lines to somewhere?
Can you give me some more details so that I can check?
Sorry if the questions are too naive.

Mgaskmg32org, 2019-06-06:
@Imtiazkarimik32 Shoot I’ll get some rn just wait

Imtiazkarimik32, 2019-07-15:
@Mgaskmg32org Hey, I tried these but couldn't make them work. Any help, please?

Mgaskmg32org, 2019-10-27:
@Imtiazkarimik23 There should be a jailbreak tweak if jailbroken...