libimobiledevice/libimobiledevice #775: Difficulty installing via Homebrew

kenny-evitt, 2019-05-07:
I was installing this library via the Homebrew package at the direction of the README for this project:

 - [NativeScript/nativescript-dev-appium: A package to help with writing and executing e2e Appium tests in NativeScript apps](https://github.com/NativeScript/nativescript-dev-appium)

I wasn't able to install *libimobiledevice* via Homebrew on my first attempt:

```
$ brew install libimobiledevice --HEAD
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core and homebrew/cask).
==> New Formulae
pipx
==> Updated Formulae
allureofthestars    awscli              camlp4              libmatroska         mariadb@10.1        mkvtoolnix          mu                  nexus               php@7.1             tor                 unp64               velero              yara
==> Deleted Formulae
imapsync

Warning: Your Xcode (10.1) is outdated.
Please update to Xcode 10.2 (or delete it).
Xcode can be updated from the App Store.

Warning: A newer Command Line Tools release is available.
Update them from Software Update in System Preferences.

==> Cloning https://git.libimobiledevice.org/libimobiledevice.git
Updating /Users/kenny/Library/Caches/Homebrew/libimobiledevice--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at eea4f1b Updated ac_python_devel.m4 to newer ax_python_devel.m4
==> ./autogen.sh
Last 15 lines from /Users/kenny/Library/Logs/Homebrew/libimobiledevice/01.autogen.sh:
checking dynamic linker characteristics... darwin18.2.0 dyld
checking how to hardcode library paths into programs... immediate
checking for pkg-config... /usr/local/opt/pkg-config/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libusbmuxd >= 1.1.0... no
configure: error: Package requirements (libusbmuxd >= 1.1.0) were not met:

Requested 'libusbmuxd >= 1.1.0' but version of libusbmuxd is 1.0.10

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libusbmuxd_CFLAGS
and libusbmuxd_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

READ THIS: https://docs.brew.sh/Troubleshooting

Error: Your Xcode (10.1) is outdated.
Please update to Xcode 10.2 (or delete it).
Xcode can be updated from the App Store.

Error: A newer Command Line Tools release is available.
Update them from Software Update in System Preferences.
```

I ended up updating macOS, Xcode, and Command Line Tools, and installing the 'HEAD' version of the *usbmuxd* Homebrew package; details:

I tried overwriting the links for *pkg-config*:

```
✔ ~/@code/partially-pos [integration-testing L|✚ 2…4] 
(cmd)14:03 $ brew link --overwrite pkg-config
Linking /usr/local/Cellar/pkg-config/0.29.2... 6 symlinks created
```

I retried installing *libimobiledevice*:
[... 196 lines cut ...]
🍺  /usr/local/Cellar/usbmuxd/HEAD-873252d_1: 13 files, 141.2KB, built in 46 seconds
Removing: /usr/local/Cellar/usbmuxd/1.0.10_1... (13 files, 120KB)
Removing: /Users/kenny/Library/Caches/Homebrew/usbmuxd--1.0.10_1.mojave.bottle.tar.gz... (38.8KB)
```

I was seemingly able to then install *libimobiledevice*:

```
✔ ~/@code/partially-pos [integration-testing L|✚ 2…4] 
(ins)09:51 $ brew install libimobiledevice --HEAD
==> Cloning https://git.libimobiledevice.org/libimobiledevice.git
Updating /Users/kenny/Library/Caches/Homebrew/libimobiledevice--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at eea4f1b Updated ac_python_devel.m4 to newer ax_python_devel.m4
==> ./autogen.sh
==> ./configure --disable-silent-rules --prefix=/usr/local/Cellar/libimobiledevice/HEAD-eea4f1b_3 --without-cython --enable-debug-code
==> make install
🍺  /usr/local/Cellar/libimobiledevice/HEAD-eea4f1b_3: 67 files, 1MB, built in 56 seconds
```

I created a GitHub Gist of my complete notes:

 - [*nativescript-dev-appium* installation notes](https://gist.github.com/kenny-evitt/4eb84acf94defdcdfceefa20e2a887b3)

nickoneill, 2019-05-08:
I also ran into this today, and the above fix worked for me as well.

cejka, 2019-05-17:
Sweet, this did it!

Start from the "I tried installing the HEAD version" part. You'll have to unlink libimobiledevice as well.

Summary:

> brew cask install osxfuse

not sure if "cask" is needed
reboot system and device just in case
>brew install ifuse
brew unlink usbmuxd
brew install usbmuxd --HEAD
brew unlink libimobiledevices
brew install libimobiledevices --HEAD

brew made me reinstall a few packages (autoconf, automake, etc) - just 'brew reinstall ..." them if it asks.

I mounted it the first time this way (where the 000s are your device's UDID)
>mkdir ~/ipad
>ifuse ~/ipad -u 0000000000000000

if it's not working, run this to see if anything shows up
>ideviceinfo

Then check the list here to figure out how to access what you need:
http://www.libimobiledevice.org/