libimobiledevice/libimobiledevice #776: OpenSSL error

Dantee296, 2019-05-10:
macos:libimobiledevice make 
Making all in common
make[2]: Nothing to be done for `all'.
Making all in src
  CC       idevice.lo
  CCLD     libimobiledevice.la
Undefined symbols for architecture x86_64:
  "_BIO_free", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_BIO_int_ctrl", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_BIO_new", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_BIO_new_mem_buf", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_BIO_s_socket", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_CRYPTO_cleanup_all_ex_data", referenced from:
      _internal_idevice_deinit in idevice.o
  "_CRYPTO_num_locks", referenced from:
      _internal_idevice_init in idevice.o
      _internal_idevice_deinit in idevice.o
  "_CRYPTO_set_id_callback", referenced from:
      _internal_idevice_init in idevice.o
      _internal_idevice_deinit in idevice.o
  "_CRYPTO_set_locking_callback", referenced from:
      _internal_idevice_init in idevice.o
      _internal_idevice_deinit in idevice.o
  "_ERR_remove_thread_state", referenced from:
      _openssl_remove_thread_state in idevice.o
  "_EVP_cleanup", referenced from:
      _internal_idevice_deinit in idevice.o
  "_PEM_read_bio_RSAPrivateKey", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_PEM_read_bio_X509", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_RSA_free", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_COMP_free_compression_methods", referenced from:
      _internal_idevice_deinit in idevice.o
  "_SSL_CTX_free", referenced from:
      _idevice_connection_enable_ssl in idevice.o
      _internal_ssl_cleanup in idevice.o
  "_SSL_CTX_new", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_CTX_use_RSAPrivateKey", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_CTX_use_certificate", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_do_handshake", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_free", referenced from:
      _idevice_connection_enable_ssl in idevice.o
      _internal_ssl_cleanup in idevice.o
  "_SSL_library_init", referenced from:
      _internal_idevice_init in idevice.o
  "_SSL_new", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_read", referenced from:
      _idevice_connection_receive_timeout in idevice.o
      _idevice_connection_receive in idevice.o
  "_SSL_set_bio", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_set_connect_state", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_set_verify", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_shutdown", referenced from:
      _idevice_connection_disable_ssl in idevice.o
  "_SSL_write", referenced from:
      _idevice_connection_send in idevice.o
  "_TLSv1_method", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_X509_free", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_asn1_array2tree", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_asn1_create_element", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_asn1_delete_structure", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_asn1_der_decoding", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_asn1_read_value", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gcry_control", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_free", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_hash_fast", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_hash_get_len", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_malloc", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_pem_base64_decode2", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_crt_deinit", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_crt_export", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_crt_import", referenced from:
      _pair_record_import_crt_with_name in libinternalcommon.a(userpref.o)
  "_gnutls_x509_crt_init", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_crt_set_activation_time", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_crt_set_ca_status", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_crt_set_expiration_time", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_crt_set_key", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_crt_set_key_usage", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_crt_set_serial", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_crt_set_subject_key_id", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_crt_set_version", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_crt_sign2", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_privkey_deinit", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_privkey_export", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_privkey_generate", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_privkey_import", referenced from:
      _pair_record_import_key_with_name in libinternalcommon.a(userpref.o)
  "_gnutls_x509_privkey_import_rsa_raw", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_privkey_init", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libimobiledevice.la] Error 1
make[1]: *** [all-recursive] Error 1

closed 2019-05-28 by Dantee296

trungtin, 2019-06-12:
Why did you close @Dantee296? please help others with a similar issue. Did you solve it?

nikias, 2019-06-12:
@trungtin this totally looks like a manually created problem there. It seems both OpenSSL and GnuTLS was enabled somehow which wouldn't work anyway.

Dantee296, 2019-06-12:
guys i wasnt able to compile via openssl what so ever only pption was using
gnutls

Dantee296, 2019-06-13:
@nikias how to disable Gnutls while compiling with OpenSSL

nikias, 2019-06-13:
If OpenSSL is detected it will disable GnuTLS automatically. You can enforce selecting OpenSSL with `--enable-OpenSSL` passing it to configure or autogen.sh and it will print `SSL support backend :  OpenSSL` accordingly. Just check the output.

Dantee296, 2019-06-13:
@nikias  So first of all openssl_CFLAGS and openssl_LIBS flags were not set and we get this error on autogen.sh
> checking for openssl >= 0.9.8... no
> configure: error: OpenSSL support explicitly requested but OpenSSL could not be found

than i set these flags 

> export openssl_LIBS="-L/usr/local/opt/openssl/lib"
> export openssl_CFLAGS="-I/usr/local/opt/openssl/include/"
and out was "checking for openssl >= 0.9.8... yes"

Configuration for libimobiledevice 1.2.1:
-------------------------------------------

  Install prefix: .........: /usr/local
  Debug code ..............: no
  Python bindings .........: no
  SSL support backend .....: OpenSSL

now on "make " i get this error

CCLD     libimobiledevice.la
Undefined symbols for architecture x86_64:
  "_BIO_free", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_BIO_int_ctrl", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_BIO_new", referenced from:
      _idevice_connection_enable_ssl in idevice.o .... and much more like 
"_gnutls_x509_privkey_export", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_SSL_read", referenced from:
      _idevice_connection_receive_timeout in idevice.o

mexmer, 2019-06-13:
if you already compiled source (with errors), first this you should do is 
make clean

then you run ./configure with new parameters you want.

if you do ./configure without make clean, there will be old compiled files artifacts, and make will not create new ones, and you will get link errors like you posted.

Dantee296, 2019-06-13:
@mexmer just tried , didnt work

Dantee296, 2019-06-13:
CCLD     libimobiledevice.la
Undefined symbols for architecture x86_64:
  "_BIO_free", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_BIO_int_ctrl", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_BIO_new", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_BIO_new_mem_buf", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_BIO_s_socket", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_CRYPTO_cleanup_all_ex_data", referenced from:
      _internal_idevice_deinit in idevice.o
  "_CRYPTO_num_locks", referenced from:
      _internal_idevice_init in idevice.o
      _internal_idevice_deinit in idevice.o
  "_CRYPTO_set_id_callback", referenced from:
      _internal_idevice_init in idevice.o
      _internal_idevice_deinit in idevice.o
  "_CRYPTO_set_locking_callback", referenced from:
      _internal_idevice_init in idevice.o
      _internal_idevice_deinit in idevice.o
  "_ERR_remove_thread_state", referenced from:
      _internal_idevice_deinit in idevice.o
      _idevice_connection_enable_ssl in idevice.o
  "_EVP_cleanup", referenced from:
      _internal_idevice_deinit in idevice.o
  "_PEM_read_bio_RSAPrivateKey", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_PEM_read_bio_X509", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_RSA_free", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_COMP_get_compression_methods", referenced from:
      _internal_idevice_deinit in idevice.o
  "_SSL_CTX_free", referenced from:
      _idevice_connection_disable_ssl in idevice.o
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_CTX_new", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_CTX_use_RSAPrivateKey", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_CTX_use_certificate", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_do_handshake", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_free", referenced from:
      _idevice_connection_disable_ssl in idevice.o
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_library_init", referenced from:
      _internal_idevice_init in idevice.o
  "_SSL_new", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_read", referenced from:
      _idevice_connection_receive_timeout in idevice.o
      _idevice_connection_receive in idevice.o
  "_SSL_set_bio", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_set_connect_state", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_set_verify", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_SSL_shutdown", referenced from:
      _idevice_connection_disable_ssl in idevice.o
  "_SSL_write", referenced from:
      _idevice_connection_send in idevice.o
  "_TLSv1_method", referenced from:
      _idevice_connection_enable_ssl in idevice.o
  "_X509_free", referenced from:
[... 43 lines cut ...]
  "_gnutls_x509_crt_set_subject_key_id", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_crt_set_version", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_crt_sign2", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_privkey_deinit", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_privkey_export", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_privkey_generate", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_privkey_import", referenced from:
      _pair_record_import_key_with_name in libinternalcommon.a(userpref.o)
  "_gnutls_x509_privkey_import_rsa_raw", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_gnutls_x509_privkey_init", referenced from:
      _pair_record_generate_keys_and_certs in libinternalcommon.a(userpref.o)
  "_sk_free", referenced from:
      _internal_idevice_deinit in idevice.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libimobiledevice.la] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

mexmer, 2019-06-13:
this means you either don't have openssl development package, or if you have made it manually, you didn't properly set path to openssl library.

Dantee296, 2019-06-14:
@mexmer  this is how i set path
export openssl_LIBS="-L/usr/local/opt/openssl/lib"
export openssl_CFLAGS="-I/usr/local/opt/openssl/include/"

and in autogen.sh i got this responce 
and out was "checking for openssl >= 0.9.8... yes"

Configuration for libimobiledevice 1.2.1:
Install prefix: .........: /usr/local
Debug code ..............: no
Python bindings .........: no
SSL support backend .....: OpenSSL

Dantee296, 2019-06-14:
openssl version -a
LibreSSL 2.6.5
built on: date not available
platform: information not available
options:  bn(64,64) rc4(ptr,int) des(idx,cisc,16,int) blowfish(idx) 
compiler: information not available
OPENSSLDIR: "/private/etc/ssl"

mexmer, 2019-06-14:
libressl is not openssl, it's something different.

anyways, from your errors it looks like if you have enabled both, openssl and tls.

i would suggest to just download libimobile source again into empty directory, and run ./configure again.
because it seems, that your current build folder is broken in more than one way.

Dantee296, 2019-06-14:
@mexmer you were right current build is broken , cloning fresh source compiled easily

RainyLave, 2024-06-07:
same question too, with every package cloned in separate folder: 
```
/usr/bin/ld: warning: libssl.so.3, needed by ../src/.libs/libimobiledevice-1.0.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libcrypto.so.3, needed by ../src/.libs/libimobiledevice-1.0.so, not found (try using -rpath or -rpath-link)
../src/.libs/libimobiledevice-1.0.so: undefined reference to `EVP_MD_CTX_free@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_mod_word@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_check_prime@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `SSL_shutdown@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `PEM_read_bio_PUBKEY@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `EVP_DecryptInit_ex@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `ASN1_INTEGER_set@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BIO_ctrl@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `EVP_PKEY_free@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `ASN1_INTEGER_free@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `EVP_chacha20_poly1305@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `X509V3_EXT_conf_nid@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `PEM_write_bio_PrivateKey@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `SSL_CTX_use_certificate@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `X509_set_version@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_mod_exp@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `X509_EXTENSION_free@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `EVP_sha1@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `EVP_DigestInit@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `SSL_free@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_sub@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `EVP_EncryptUpdate@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BIO_s_mem@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_mod_exp_mont@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `SSL_CTX_ctrl@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_bn2hex@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BIO_new_mem_buf@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_div@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `SSL_CTX_free@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_CTX_new@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_MONT_CTX_free@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_lshift@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `SSL_get_error@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_mul@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `SSL_do_handshake@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_get_word@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `SSL_set_connect_state@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BIO_s_null@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `RAND_bytes@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `X509_set_serialNumber@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `X509_set1_notAfter@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_mod_mul@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `EVP_MD_get_size@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_mul_word@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `RAND_status@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `EVP_CIPHER_CTX_new@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `EVP_DigestFinal@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_MONT_CTX_set@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_num_bits@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `ASN1_TIME_free@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `EVP_sha512@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `X509_add_ext@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `SSL_set_shutdown@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `ASN1_TIME_set@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `ASN1_TIME_new@OPENSSL_3.0.0'
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BIO_set_callback_ex@OPENSSL_3.0.0'
[... cut ...]
../src/.libs/libimobiledevice-1.0.so: undefined reference to `BN_clear_free@OPENSSL_3.0.0'
collect2: error: ld returned 1 exit status
Makefile:741: recipe for target 'idevice_id' failed
make[2]: *** [idevice_id] Error 1
make[2]: Leaving directory '/home/rainy/libimobiledevice/tools'
Makefile:464: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/rainy/libimobiledevice'
Makefile:394: recipe for target 'all' failed
make: *** [all] Error 2```

RainyLave, 2024-06-07:
I use OpenSSL 1.1.1  11 Sep 2018
is this version of openssl dismatch with it?

RainyLave, 2024-06-07:
OK, I succeeded with these instructions:
ln -s /usr/local/openssl/lib/libssl.so.3 /usr/lib/libssl.so.3
ln -s /usr/local/openssl/lib64/libssl.so.3 /usr/lib64/libssl.so.3
ln -s /usr/local/openssl/lib/libcrypto.so.3 /usr/lib/libcrypto.so.3
ln -s /usr/local/openssl/lib64/libcrypto.so.3 /usr/lib64/libcrypto.so.3

RainyLave, 2024-06-07:
error then, when I install usbmuxd:
```
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libusb... yes
checking for libplist... yes
checking for libimobiledevice... no
checking for limd_glue... yes
configure: error: preflight worker support requested but libimobiledevice could not be found
rainy@ubuntu:~/u
```

RainyLave, 2024-06-07:
should i use: export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib ?
but this does't work too

RainyLave, 2024-06-07:
[image]
[image]
but it seems that libimobiledevice has been installed correctly~
oh~ :-(

RainyLave, 2024-06-08:
OK, 
```rainy@ubuntu:~/libimobiledevice$ PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh```
works!
by the way, we can use it in a more efficent way:
just copy the whole *.pc files into /usr/lib/pkhconfig
just like this:
[image]

RainyLave, 2024-06-08:
it works well!
[image]

sujona, 2024-11-07:
I managed to get past the 

```
checking for openssl >= 0.9.8... no
configure: error: OpenSSL support explicitly requested but OpenSSL could not be found

```
by installing 

`sudo apt-get install libssl-dev`