libimobiledevice/libimobiledevice #778: diagnostics_relay_query_ioregistry_entry(diagnostics_client, "ASPStorage", "", &node) return node==NULL

bugHappy, 2019-05-13:
call  diagnostics_relay_query_ioregistry_entry(diagnostics_client, "ASPStorage", "", &node)
this API return DIAGNOSTICS_RELAY_E_SUCCESS but node's value is NULL
my ios ver 9.0.2...

Kanong017, 2019-05-18:
call diagnostics_relay_query_ioregistry_entry(diagnostics_client, "", "ASPStorage", &node)

bugHappy, 2019-05-21:
diagnostics_relay_query_ioregistry_entry(diagnostics_client, "", "ASPStorage", &node) 和diagnostics_relay_query_ioregistry_entry(diagnostics_client, "", "", &node) 的结果是一样。。感觉第二个参数根本没用。第二个参数不管写什么都是返回一样的。

bugHappy, 2019-05-22:
@Kanong017 diagnostics_relay_query_ioregistry_entry(diagnostics_client, "", "ASPStorage", &node) 和diagnostics_relay_query_ioregistry_entry(diagnostics_client, "", "", &node) 的结果是一样。。感觉第二个参数根本没用。第二个参数不管写什么都是返回一样的。

closed 2019-06-21 by bugHappy