libimobiledevice/libimobiledevice #783: Connection under cygwin

mexmer, 2019-05-22:
Hello,
i have compiled libplist, libusmuxd, libimobiledevice under cygwin on windows 1810, and i'm running latest version of itunes.

for some reason non of tools connects to any ios device connected to my computer. tbh. it seem, it can't even connect to itunes mux

i compiled libimobile with --enabled-debug-code, but it doesn't give more information

```
$ ideviceinfo --debug
socket_connect: getaddrinfo: Unknown error
No device found, is it plugged in?
```

nikias, 2019-05-22:
Ok it might be something wrong with the win32 socket code. Can you make sure, in libusbmuxd, in funcion socket_connect: https://github.com/libimobiledevice/libusbmuxd/blob/master/common/socket.c#L242
WSAStartup is being called? Just add a printf before it. Also make sure WIN32 is defined.

mexmer, 2019-05-22:
it was not hit it seems.
```
$ ideviceinfo -d
socket_connect(): No error
socket_connect() not win32: No error
socket_connect: getaddrinfo: Unknown error
No device found, is it plugged in?
```

i put in 3 perror lines
1st before #ifdef WIN32, 2nd inside WIN32 block, and 3rd after else
1st and 3rd are printed

looks like WIN32 is not defined.

is there any configure switch to enable it?

i'm using cygwin64 bit edition

mexmer, 2019-05-22:
guess some problem with host detection?

```
checking build system type... x86_64-unknown-cygwin
checking host system type... x86_64-unknown-cygwin
```

mexmer, 2019-05-22:
althought when i think about it, should cygwin require WIN32 define?
i mean, it has own posix like socket implementation, so it should not require WSA

mexmer, 2019-05-22:
this happens, when i add 
#define WIN32 to config.h
[make.out.txt](https://github.com/libimobiledevice/libimobiledevice/files/3208102/make.out.txt)

nikias, 2019-05-22:
Right... cygwin should handle it but then again I am not sure. I will check it.

mexmer, 2019-05-23:
thanks, if you need more info let me know.
i want to see if libimobile works under windows, because my code(that uses WSA), returns 5 on usbmux connection, so i'm not able even list devices.

nikias, 2019-05-23:
Yeah cygwin is really messed up for me. I know it compiles perfectly fine using MinGW and then also you won't have the cygwin1.dll dependency.

mexmer, 2019-05-24:
i will try mingw, but cygwin has much easier setup.

mexmer, 2019-05-28:
what version of mingw you using? i'm unable to get correct version, that contains everything needed.
one with gui installer is missing python, also having old tools (like automake 1.11 and so on)

there is also mingw binary, which seems to be actually, but doesn't have msys shell

and there supposed to be msys2, but their repo is unreachable, so i cannot get that one.

nikias, 2019-05-28:
Oh I always compile without python because this always failed for me hehe. I have an old version in a Windows XP VM that works well for all I need at least for 32 bit.

mexmer, 2019-05-28:
hmm, idk how to deal with this
```
./configure: line 15925: syntax error near unexpected token `libplist,'
./configure: line 15925: `PKG_CHECK_MODULES(libplist, libplist >= $LIBPLIST_VERSION)'
```
in cygwin
export PKG_CONFIG_PATH=/usr/local/lib/pkconfig
was enough, but it doesn't seem to help on mingw

nikias, 2019-05-28:
For me for some reason it ends up to be in /local/lib so I do:
```
PKG_CONFIG_PATH=/local/lib/pkgconfig ./autogen.sh
```

mexmer, 2019-05-28:
nah, this seems to be ok this way, but problem was missing pkg-config, there is no package for that in mingw i have installed. i took pkg-config (and pkg.m4) from gtk, so i will let you know, when i finally compile this thing.

mexmer, 2019-05-28:
any way to set openssl path in libimobile configure? i had found only disable-openssl, but since there is no gnutls in mingw, and no openssl-dev package, i ended up compiling openssl myself, but it's installed in /usr/local and configure can't find it.

mexmer, 2019-05-29:
i'm giving up i guess. i spent lot of time on this, and only result i get is 

```
$ ideviceinfo.exe -d
No device found, is it plugged in?
```

i have run ./configure --enable-debug-code, when compiling libimobile, but there doesn't seem to be any debug info going out

mexmer, 2019-05-29:
ok, seems i needed to restart apple service and replug ipad, now it's working.

mexmer, 2019-05-29:
so, you need to install msys2, openssl-devel, mingw64 toolchain and automake and other stuff.

closed 2019-05-29 by mexmer