libimobiledevice/libimobiledevice #786: idevicebackup2 can not restore files successfully

Alisabb, 2019-05-28:
idevicebackup2 can not restore files successfully?Although the prompt restore succeeded，the restored files do not appear on the phone.

nikias, 2019-05-28:
It depends on the files; there are a few options to consider, usually a real restore requires
```
idevicebackup2 --system --settings --reboot restore BACKUPDIR
```
If the device is not rebooted after restore the files are not moved into place.

nikias, 2019-05-28:
Note that I changed the code already a while ago so --reboot is not required anymore, unless you are using an old version. Just make sure the device is rebooted when restore is complete.

Alisabb, 2019-05-29:
thank you very much ,it is successful now

closed 2019-05-30 by nikias