libimobiledevice/libimobiledevice #787: Restore for the second time? it is failed?why?

Alisabb, 2019-05-28:
ErrorCode 102: Failed to move the incomplete restore directory into place at path "/var/.backup.i" (MBErrorDomain/102). Underlying error: 鈥?backup.i鈥?couldn鈥檛 be moved to 鈥渧ar鈥?because an item with the same name already exists. at path "/var/.backup.i" (NSCocoaErrorDomain/516). Underlying error: The operation couldn鈥檛 be completed. File exists (NSPOSIXErrorDomain/17)..
Restore Failed (Error Code 102).

Blefish, 2019-05-28:
Did you do a reboot after backup restore?

nikias, 2019-05-28:
The above happens when you restore a backup without --reboot option, and restore again. I guess it is time to flip the options to make reboot default, and make it  a --no-reboot option instead.

nikias, 2019-05-28:
Actually I already changed the option to --no-reboot and forgot about it. Are you using an older version maybe?
In any case, make sure to reboot the device after restore.

Alisabb, 2019-05-29:
ok,thank you very much,and now it works.

closed 2019-05-30 by nikias