libimobiledevice/libimobiledevice #789: anyone which key use to fetch Applecare+ warranty status via using libimobiledevice

Johnny314, 2019-05-30:
i need to pick applecare+ warranty status via libimobiledevice but does not know the service or key which will return the applecare+ plist @nikias  @FunkyM  please advise

nikias, 2019-05-30:
To my knowledge there is no such information available and I would be surprised to find it. The device also does not know anything about its own warranty status. Only Apple does.

Mgaskmg32org, 2019-06-06:
@nikias Any device with iOS 12.3 (I think or 12.4) or above will show warranty status.

If warranty expires device will not show warranty status anymore.

Johnny314, 2019-07-26:
@nikias  @Mgaskmg32org  any updates regarding this issue?

Mgaskmg32org, 2019-10-27:
@Johnny314 No. Sorry!

Dantee296, 2020-04-24:
its not possible via lockdown , need 3rd party gsx api's