libimobiledevice/libimobiledevice #790: ideviceinfo: Fetch EID for ESIM slots and CSN number for iPad

nemecence90, 2019-05-31:
Hey @nikias, i am trying to find out EID related to ESIMs specifically and CSN numberr for iPads, i couldnt find anything except 2 IMEIs for devices with ESIM,

> CarrierBundleInfoArray[1]: 
>  0: 
>   CFBundleIdentifier: com.apple.ATT_US
>   CFBundleVersion: 36.0
>   GID1: ffff
>   GID2: ffff
>   IntegratedCircuitCardIdentity: 89014104XXXXXXXXXX
>   InternationalMobileSubscriberIdentity: 310XXXXXXXXXX
>   MCC: 310
>   MNC: 410
>   MobileEquipmentIdentifier: 356XXXXXXXX
>   SIMGID1: //8=
>   SIMGID2: //8=
>   Slot: kOne
>   kCTPostponementInfoAvailable: SIMCarrierInfo
> CertID: 165XXXXX
> ChipID: 32800
> ChipSerialNo: dTOnXXXXXX
> DeviceClass: iPhone
> DeviceColor: 1
> DeviceName: iPhone (2)
> DieID: 31XXXXXXXXX
> EthernetAddress: 38:XX:XX:XX:XX:65
> FirmwareVersion: iBoot-4513.250.288
> FusingStatus: 3
> GID1: ffff
> GID2: ffff
> HardwareModel: D321AP
> HardwarePlatform: t8020
> HasSiDP: true
> HostAttached: true
> IntegratedCircuitCardIdentity: 890XXXXXXXXXXXXXXXX
> InternationalMobileEquipmentIdentity: 35XXXXXXXXXXXXXXXX
> InternationalMobileEquipmentIdentity2: 35XXXXXXXXXXXXXXXX
> InternationalMobileSubscriberIdentity: 310XXXXXXXXXXXXXXXX
> InternationalMobileSubscriberIdentityOverride: false
> MLBSerialNumber: G2CXXXXXXXXXXXXXXXX
> MobileEquipmentIdentifier: 356XXXXXXXXXXXXXXXX
> MobileSubscriberCountryCode: 
> MobileSubscriberNetworkCode: 410

I cannot see any other data related to ESIM specifically, and also is it possible to get CSN number for iPads?

nikias, 2019-09-18:
I don't really know, sorry. It might not be accessible via USB. Is there any hint that iTunes can get it?

truonggiang0710, 2021-12-14:
Hi @uncommonthinker , do you find out any solution?

truonggiang0710, 2022-01-20:
HI @uncommonthinker , do you have any updates?

uncommonthinker, 2022-01-20:
Hey @truonggiang0710 , I have not checked it recently. I was not following up on this. Will let you know if I find anything.

uncommonthinker, 2022-01-20:
Hey @nikias, Sorry for the late reply. 
I cannot see the EID anywhere in the Finder and Music App but the iPhone settings does show it. Attaching the image.
[image]

mexmer, 2022-01-20:
you can check mobilegestalt, some information are not in deviceinfo, and it's not limitation of libimobile, libimobile asks for all values and apple returns, what he wants.

truonggiang0710, 2022-01-21:
Hi @mexmer , I checked with key "MobileEquipmentInfoCSN", "MobileEquipmentInfoDisplayCSN", the command as
` idevicediagnostics mobilegestalt MobileEquipmentInfoCSN`
` idevicediagnostics mobilegestalt MobileEquipmentInfoDisplayCSN`
The plist returned successfully, but the plist does not contain any values of CSN in the plist.
Do you know which key to get CSN/EID?

mexmer, 2022-01-24:
unfortunately i have no idea, but you can list all mobilegestalt keys and their values, and search trough that.

truonggiang0710, 2022-01-25:
I will try to do that. Thanks @mexmer

sriteja777, 2023-01-24:
Hi @uncommonthinker @truonggiang0710 were you able to find any thing over here to get EID?