libimobiledevice/libimobiledevice #794: problem in updating iOS13

UniqueLegend, 2019-06-05:
I successfully updated my iPad Pro(10.5') yesterday with iPadOS-13 ipsw. Today I try to use the same method  to update my iPhone6s:

`> idevicerestore.exe -d ios13.ipsw`

but it goes repeatedly like 
`
FDR 0000000004CFE910 timeout waiting for command 
FDR 0000000004CFE910 waiting for message...
No data to read
`

After it finished, my iphone ran into recovery mode...
[image]

Mgaskmg32org, 2019-06-06:
Hey, sorry but to say but it needs to be restored. This happened to me too but I had a backup.
I don’t know if it’s going to be the same story for you.
At the end I had to use a Mac with MacOS Catalina. Then I got iOS 13

Mgaskmg32org, 2019-06-06:
Your ECID was reset too so I can’t fix that...

UniqueLegend, 2019-06-08:
> Your ECID was reset too so I can’t fix that...
> But the Recovery fixes it
> 
> Does your phone still turn on and work?

Yes it works well. btw thks for developing this tool :)

closed 2019-06-15 by UniqueLegend

nikias, 2019-06-15:
FYI the messages
```
FDR 00000000******** waiting for message...
FDR 00000000******** timeout waiting for command
```
Can be safely ignored, those are not indicating any error at all.