libimobiledevice/libimobiledevice #795: Warning for people who tried restoring with iOS 13 IPSW (Me)

Mgaskmg32org, 2019-06-06:
When using the command prompt on Win10!

Use command prompt to restore iOS 13 IPSW
When applying all the correct commands
the restore will start.
At the end the command prompt will give error
Delete/uninstall all Apple iBoot driver from Device manager.
Give Windows time to reinstall the driver
Then start command again
The command prompt will start restore

End (for me)
Error will pop up saying 000000XXXXXX
X would be all the numbers at the end
But the 0s mean the ECID was reset
Note: That’s what sucked for me because took me 9-14 Hours to fix
Then at the end I used a Mac with MacOS Catalina and updated will IPSW HAPPY
iOS 13 I’m not going to link this but I am:
betaprofiles.com (Good guy)

I tried this and at the end the Device’s
(iPhone 8 Plus) ECID was reset and no program can scan the ECID and gives error saying 
“Can’t read ECID”

I fixed it by Restoring on iTunes:
(FOR iPhone 7 and Up)
(Do all fast)
Volume up (Let go)
Volume down (Let go again)
Hold Sleep/Wake/Side button
Restore will pop-up

(FOR iPhone 6s Plus and Below)
Power + Home (Hold)
Device screen will go dark
When AppleLogo appears>Release Home
Restore will pop-up

mexmer, 2019-06-07:
if it's official ipsw, you can upload it to device with itunes.

Mgaskmg32org, 2019-06-07:
Yes, but I don’t think iTunes with Win or Mac OS Mojave has compatibility with a iOS 13 Restore IPSW

mexmer, 2019-06-07:
restoring IPSW only does validation of image signature against apple server, process itself is done by phone. itunes uploads ipsw into phone, then it restarts, and rest is done by phone, same way libimobile does it too.

nikias, 2019-06-15:
I don't really know what is going on there but
a) This ticket was posted in the wrong project
b) Windows is not officially supported and tested (even though it should work but also might horribly fail)
c) ECID is the hardware ID (also called Chip ID) of the device and I am pretty sure it can't be deleted or erased or reset in any way
d) Recovery/DFU Mode restore is also possible with idevicerestore

closed 2019-10-27 by Mgaskmg32org