libimobiledevice/libimobiledevice #801: Restore Failed (Error Code 205).

Hiepars, 2019-06-14:
Sending '294ed6e773c1755b352357480bc69d94b687db43/Status.plist' (189 Bytes)
Sending '294ed6e773c1755b352357480bc69d94b687db43/Manifest.plist' (22.7 KB)
Sending '294ed6e773c1755b352357480bc69d94b687db43/Manifest.db' (1.4 MB)
Sending '294ed6e773c1755b352357480bc69d94b687db43/Info.plist' (100.8 KB)

[====                                              ]   6% Finished
ErrorCode 205: Multiple errors downloading files (MBErrorDomain/205)

thanks for your help.

mexmer, 2019-06-14:
either backup process could not read files from disk, or it could not write files to phone

make sure, there is enough free space in phone, before doing restore.

Hiepars, 2019-06-14:
> either backup process could not read files from disk, or it could not write files to phone
> 
> make sure, there is enough free space in phone, before doing restore.

this is restore process issue: Restore Failed (Error Code 205)
thanks for your answer.

closed 2019-06-19 by Hiepars