libimobiledevice/libimobiledevice #802: idevicesyslog fails to display logs

machoreverser, 2019-06-15:
So on mojave 10.14.5 and iOS 12.1.2, I get the following when I run idevicesyslog:

```
brew install --HEAD libimobiledevice
==> Cloning https://git.libimobiledevice.org/libimobiledevice.git
Updating /Users/machoreverser/Library/Caches/Homebrew/libimobiledevice--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 7c00b36 debugserver: Fix for iOS 13
==> ./autogen.sh
==> ./configure --disable-silent-rules --prefix=/usr/local/Cellar/libimobiledevice/HEAD-7c00b36_3 --without-cython --enable-debug-code
==> make install
 /usr/local/Cellar/libimobiledevice/HEAD-7c00b36_3: 67 files, 1MB, built in 34 seconds
```

When I run idevicesyslog -d I get:
```
---
</plist>
13:26:28 lockdown.c:303 lockdownd_stop_session(): success
13:26:28 idevice.c:900 idevice_connection_disable_ssl(): SSL mode disabled
13:26:28 syslog_relay.c:73 syslog_relay_client_new(): Creating syslog_relay_client, port = 53531.
13:26:28 syslog_relay.c:88 syslog_relay_client_new(): syslog_relay_client successfully created.
[connected]
13:26:28 syslog_relay.c:150 syslog_relay_worker(): Running
13:26:28 idevice.c:393 socket_recv_to_idevice_error(): ERROR: received timeout (Operation timed out)
13:26:28 service.c:167 service_receive_with_timeout(): could not read data
13:26:28 syslog_relay.c:133 syslog_relay_receive_with_timeout(): Could not read data, error -256
13:26:28 syslog_relay.c:159 syslog_relay_worker(): Connection to syslog relay interrupted
13:26:28 syslog_relay.c:171 syslog_relay_worker(): Exiting
```
Did I miss something? Is this supported on iOS 12?

nikias, 2019-06-15:
Hi, this was recently introduced while trying to fix an issue with iOS 13. Pull latest code (7a0ab5f8f25b1c7f0c7313d7feda9c41c8058702) and try again!

machoreverser, 2019-06-16:
All fixed. Appreciated!

closed 2019-06-16 by machoreverser