libimobiledevice/libimobiledevice #803: ios 13 issues

rcmpayne, 2019-06-17:
After git pull and rebuilding everything using msys2  I am seeing issues with lockdown -5

`cd ~/libplist

./autogen.sh --without-cython
make install
cd ~/libusbmuxd
./autogen.sh
make install
cd ~/libimobiledevice
./autogen.sh --enable-debug-code --without-cython
make install
`

`C:\MDM_Device_Toolkit\Toolkit>idevice_id.exe -l
d1****dbab6a832*****c55d050c9a5a****

C:\MDM_Device_Toolkit\Toolkit>ideviceinfo.exe
ERROR: Could not connect to lockdownd, error code -5

C:\MDM_Device_Toolkit\Toolkit>idevicepair.exe unpair
SUCCESS: Unpaired with device d1****dbab6a832*****c55d050c9a5a****

C:\MDM_Device_Toolkit\Toolkit>idevicepair.exe pair
SUCCESS: Paired with device d1****dbab6a832*****c55d050c9a5a****

C:\MDM_Device_Toolkit\Toolkit>ideviceinfo.exe
ERROR: Could not connect to lockdownd, error code -5
`

`
C:\MDM_Device_Toolkit\Toolkit>ideviceinfo.exe -d
18:05:05 lockdown.c:676 lockdownd_client_new(): device udid: d16c1f7dbab6a8325f06fe6dc55d050c9a5a6a15
18:05:05 lockdown.c:405 lockdownd_query_type(): called
18:05:05 property_list_service.c:132 internal_plist_send(): sending 284 bytes
18:05:05 service.c:144 service_send(): sending 4 bytes
18:05:05 service.c:144 service_send(): sending 284 bytes
18:05:05 property_list_service.c:137 internal_plist_send(): sent 284 bytes
18:05:05 property_list_service.c:138 internal_plist_send(): printing 284 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>ideviceinfo</string>
        <key>Request</key>
        <string>QueryType</string>
</dict>
</plist>
18:05:05 property_list_service.c:200 internal_plist_receive_timeout(): initial read=4
18:05:05 property_list_service.c:206 internal_plist_receive_timeout(): 297 bytes following
18:05:05 property_list_service.c:219 internal_plist_receive_timeout(): received 297 bytes
18:05:05 property_list_service.c:245 internal_plist_receive_timeout(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Request</key>
        <string>QueryType</string>
        <key>Type</key>
        <string>com.apple.mobile.lockdown</string>
</dict>
</plist>
18:05:05 lockdown.c:421 lockdownd_query_type(): success with type com.apple.mobile.lockdown
18:05:05 property_list_service.c:132 internal_plist_send(): sending 332 bytes
18:05:05 service.c:144 service_send(): sending 4 bytes
18:05:05 service.c:144 service_send(): sending 332 bytes
18:05:05 property_list_service.c:137 internal_plist_send(): sent 332 bytes
18:05:05 property_list_service.c:138 internal_plist_send(): printing 332 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>ideviceinfo</string>
        <key>Key</key>
        <string>ProductVersion</string>
        <key>Request</key>
        <string>GetValue</string>
</dict>
</plist>
18:05:05 property_list_service.c:200 internal_plist_receive_timeout(): initial read=4
18:05:05 property_list_service.c:206 internal_plist_receive_timeout(): 325 bytes following
18:05:05 property_list_service.c:219 internal_plist_receive_timeout(): received 325 bytes
18:05:05 property_list_service.c:245 internal_plist_receive_timeout(): printing 325 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Key</key>
        <string>ProductVersion</string>
        <key>Request</key>
        <string>GetValue</string>
        <key>Value</key>
        <string>13.0</string>
</dict>
</plist>
18:05:05 lockdown.c:474 lockdownd_get_value(): success
18:05:05 lockdown.c:485 lockdownd_get_value(): has a value
18:05:05 userpref.c:187 userpref_read_system_buid(): using 307142892271866102839213504 as SystemBUID
18:05:05 property_list_service.c:132 internal_plist_send(): sending 430 bytes
18:05:05 service.c:144 service_send(): sending 4 bytes
18:05:05 service.c:144 service_send(): sending 430 bytes
18:05:05 property_list_service.c:137 internal_plist_send(): sent 430 bytes
18:05:05 property_list_service.c:138 internal_plist_send(): printing 430 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>ideviceinfo</string>
        <key>Request</key>
        <string>StartSession</string>
        <key>HostID</key>
        <string>7D7AB77F-007B-589B-2B65-672912374631</string>
        <key>SystemBUID</key>
        <string>307142892271866102839213504</string>
</dict>
</plist>
18:05:05 property_list_service.c:200 internal_plist_receive_timeout(): initial read=4
18:05:05 property_list_service.c:206 internal_plist_receive_timeout(): 354 bytes following
18:05:05 property_list_service.c:219 internal_plist_receive_timeout(): received 354 bytes
18:05:05 property_list_service.c:245 internal_plist_receive_timeout(): printing 354 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>EnableSessionSSL</key>
        <true/>
        <key>Request</key>
        <string>StartSession</string>
        <key>SessionID</key>
        <string>9A96C411-82D6-4D21-9749-A99B390E8EA8</string>
</dict>
</plist>
18:05:05 lockdown.c:1193 lockdownd_start_session(): Session startup OK
18:05:05 lockdown.c:1205 lockdownd_start_session(): SessionID: 9A96C411-82D6-4D21-9749-A99B390E8EA8
18:05:05 lockdown.c:1212 lockdownd_start_session(): Enable SSL Session: true
18:05:05 idevice.c:772 idevice_connection_enable_ssl(): WARNING: Could not load RootCertificate
**18:05:05 idevice.c:800 idevice_connection_enable_ssl(): ERROR in SSL_do_handshake: SSL_ERROR_ZERO_RETURN**
18:05:05 lockdown.c:767 lockdownd_client_new_with_handshake(): Session opening failed.
18:05:05 lockdown.c:287 lockdownd_stop_session(): stopping session 9A96C411-82D6-4D21-9749-A99B390E8EA8
18:05:05 property_list_service.c:132 internal_plist_send(): sending 363 bytes
18:05:05 service.c:144 service_send(): sending 4 bytes
18:05:05 service.c:144 service_send(): sending 363 bytes
18:05:05 idevice.c:351 internal_connection_send(): **ERROR**: usbmuxd_send returned -2 (No such file or directory)
18:05:05 service.c:147 service_send(): **ERROR**: sending to device failed.
18:05:05 property_list_service.c:147 internal_plist_send(): **ERROR**: sending to device failed.
18:05:05 idevice.c:417 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -2 (No such file or directory)
18:05:05 service.c:167 service_receive_with_timeout(): could not read data
18:05:05 property_list_service.c:196 internal_plist_receive_timeout(): initial read failed!
**18:05:05 lockdown.c:297 lockdownd_stop_session(): LOCKDOWN_E_PLIST_ERROR**
**ERROR: Could not connect to lockdownd, error code -5**
`

The errors only occure when i start playing with idevicepair.exe unpair and idevicepair.exe pair

mexmer, 2019-06-18:
this looks like source of problem
not sure, why it didn't generate root certificate for you

18:05:05 idevice.c:772 idevice_connection_enable_ssl(): WARNING: Could not load RootCertificate

just disconnect iphone, stop apple mobile device service

go to 
c:\ProgramData\Apple\Lockdown\ 
and delete all files there (mind this will remove pairing information for all ios devices connected to your computer)
start apple mobile device service again, then connect phone and do pairing (don't forget to confirm trust).

plist files with pairing info should be recreated.

nikias, 2019-06-18:
Deleting just the one with device id is usually enough. But also since usbmuxd is handling the pairing the question is why it didn't make a new pairing when the device was plugged in?

rcmpayne, 2019-06-18:
Stopped the service, deleted the *.plist files, started service and connected device and tried the commands again and same issue.

1. Stopped the service
2. Deleted the *.plist files
3. Started service
4. Connected device and trust the connection
5. Saw that the *.plist files was re-created
6. ideviceinfo.exe worked
7. idevicepair.exe unpair
7. idevicepair.exe pair
8. ideviceinfo.exe 
ERROR: Could not connect to lockdownd, error code -5
```
09:27:07 lockdown.c:1193 lockdownd_start_session(): Session startup OK
09:27:07 lockdown.c:1205 lockdownd_start_session(): SessionID: 72D06D15-4F18-48C5-8042-7EFD5608D3EA
09:27:07 lockdown.c:1212 lockdownd_start_session(): Enable SSL Session: true
09:27:07 idevice.c:772 idevice_connection_enable_ssl(): WARNING: Could not load RootCertificate
09:27:07 idevice.c:800 idevice_connection_enable_ssl(): ERROR in SSL_do_handshake: SSL_ERROR_ZERO_RETURN
09:27:07 lockdown.c:767 lockdownd_client_new_with_handshake(): Session opening failed.
09:27:07 lockdown.c:287 lockdownd_stop_session(): stopping session 72D06D15-4F18-48C5-8042-7EFD5608D3EA
09:27:07 property_list_service.c:132 internal_plist_send(): sending 363 bytes
09:27:07 service.c:144 service_send(): sending 4 bytes
09:27:07 service.c:144 service_send(): sending 363 bytes
09:27:07 idevice.c:351 internal_connection_send(): ERROR: usbmuxd_send returned -2 (No such file or directory)
09:27:07 service.c:147 service_send(): ERROR: sending to device failed.
09:27:07 property_list_service.c:147 internal_plist_send(): ERROR: sending to device failed.
09:27:07 idevice.c:417 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -2 (No such file or directory)
09:27:07 service.c:167 service_receive_with_timeout(): could not read data
09:27:07 property_list_service.c:196 internal_plist_receive_timeout(): initial read failed!
09:27:07 lockdown.c:297 lockdownd_stop_session(): LOCKDOWN_E_PLIST_ERROR
ERROR: Could not connect to lockdownd, error code -5
```
9. Disconnect and re-connect phone
8. ideviceinfo.exe  works

So unless i start telling people to disconnect phone, I still cant recover from a unpair

rcmpayne, 2019-06-18:
Switching my code to use validate vs pair as a workaround for now

```
:Attempttopairdevice
   rem This Checks the device to ensure its linked to PC
   rem ERROR -- device needs to be paired
   rem SUCCESS -- device paired successfully
set n=0
:: TODO new code has this broken. need to fix 
:: Change pair to validate
for /F %%a in ('%iOSTools%\idevuicepair.exe validate ^| %Tools%\awk.exe -F "(:)" "{print $1}"') do (
   set /A n+=1
   set "device[!n!]=%%a"
)
echo.
IF %device[1]%==ERROR (
call :colorPrint 0c "Please accept the trust dialog on the screen of device, then attempt to pair again"
echo(
echo Press any key to try again
pause > nul
goto :Attempttopairdevice
)
echo.
IF %device[1]%==No (
call :colorPrint 0c "No device found, is it plugged in?"
echo(
echo Press any key to try again
pause > nul
goto :home
)
```

nikias, 2019-06-18:
Why do you need to unpair the device if you pair it right away again?

rcmpayne, 2019-06-18:
You don't need a unpair to see this issue. I was just adding it for clarity.

nikias, 2019-06-18:
So basically `idevicepair pair` creates an invalid pairing record apparently. Mind sharing the paring record that is generated after calling `idevicepair pair` ? Maybe the tool should check if the device is already paired (and that's why it might fail).

Dantee296, 2019-09-16:
@nikias @rcmpayne facing exact same issue
itunes trust works
but as soon as u trust with idevicepair , it causing "ERROR: Could not connect to lockdownd, error code -5"

using msys2

Dantee296, 2019-09-16:
@nikias  pair records
[85335cb60ed6000491ff6e642a4ec6bd4b6abc19.plist.zip](https://github.com/libimobiledevice/libimobiledevice/files/3615035/85335cb60ed6000491ff6e642a4ec6bd4b6abc19.plist.zip)

nikias, 2019-09-16:
Can you add this line into src/idevice.c:
```
         if (SSL_CTX_use_certificate(ssl_ctx, rootCert) != 1) {
                 debug_info("WARNING: Could not load RootCertificate");
+                ERR_print_errors_fp(stderr);
         }
```

Also in common/userpref.c:

```
 #ifdef HAVE_OPENSSL
 userpref_error_t pair_record_import_crt_with_name(plist_t pair_record, const char* name, key_data_t* cert)
 #else
userpref_error_t pair_record_import_crt_with_name(plist_t pair_record, const char* name, 
 gnutls_x509_crt_t cert)
 #endif
 {
 #ifdef HAVE_OPENSSL
         if (!cert)
                 return USERPREF_E_SUCCESS;
 #endif
         userpref_error_t ret = USERPREF_E_INVALID_CONF;

 #ifdef HAVE_OPENSSL
                 ret = pair_record_get_item_as_key_data(pair_record, name, cert);
+                debug_info("ret = %d, length = %d", ret, cert->size);
 #else
```

And paste the error output?

Dantee296, 2019-09-16:
on it sir

Dantee296, 2019-09-16:
20680:error:0D07209B:asn1 encoding routines:ASN1_get_object:too long:../openssl-1.1.1b/crypto/asn1/asn1_lib.c:91:
20680:error:0D068066:asn1 encoding routines:asn1_check_tlen:bad object header:../openssl-1.1.1b/crypto/asn1/tasn_dec.c:1118:
20680:error:0D07803A:asn1 encoding routines:asn1_item_embed_d2i:nested asn1 error:../openssl-1.1.1b/crypto/asn1/tasn_dec.c:190:Type=ASN1_TIME
20680:error:0D08303A:asn1 encoding routines:asn1_template_noexp_d2i:nested asn1 error:../openssl-1.1.1b/crypto/asn1/tasn_dec.c:627:Field=notBefore, Type=X509_VAL
20680:error:0D08303A:asn1 encoding routines:asn1_template_noexp_d2i:nested asn1 error:../openssl-1.1.1b/crypto/asn1/tasn_dec.c:627:Field=validity, Type=X509_CINF
20680:error:0D08303A:asn1 encoding routines:asn1_template_noexp_d2i:nested asn1 error:../openssl-1.1.1b/crypto/asn1/tasn_dec.c:627:Field=cert_info, Type=X509
20680:error:0906700D:PEM routines:PEM_ASN1_read_bio:ASN1 lib:../openssl-1.1.1b/crypto/pem/pem_oth.c:33:
20680:error:140AB043:SSL routines:SSL_CTX_use_certificate:passed a null parameter:../openssl-1.1.1b/ssl/ssl_rsa.c:305:
ERROR: Could not connect to lockdownd, error code -5

nikias, 2019-09-16:
So according to this error message:
```
20680:error:140AB043:SSL routines:SSL_CTX_use_certificate:passed a null parameter:../openssl-1.1.1b/ssl/ssl_rsa.c:305:
```
The value of `rootCert` is NULL. What is the output when you run with `-d`? It should have the additional line from common/userpref.c

Dantee296, 2019-09-16:
@nikias  not going to that line

userpref_error_t pair_record_import_crt_with_name(plist_t pair_record, const char* name, gnutls_x509_crt_t cert)
#endif
{
//  code return from here and not going further
#ifdef HAVE_OPENSSL

nikias, 2019-09-16:
So you mean the function is not called at all?

Dantee296, 2019-09-16:
called this function
#else
userpref_error_t pair_record_import_crt_with_name(plist_t pair_record, const char* name, gnutls_x509_crt_t cert)

but seems like #ifdef HAVE_OPENSSL is returning false

Dantee296, 2019-09-16:
@nikias  i removed this check
#ifdef HAVE_OPENSSL

than got 
ret = 0, length = 912

nikias, 2019-09-16:
if that's is the case you have configured libimobiledevice wrong. `#ifdef HAVE_OPENSSL` is defined in `config.h` (which gets generated by autoconf).

nikias, 2019-09-16:
Also `#ifdef HAVE_OPENSSL` must be defined at least for idevice.c, otherwise you wouldn't get the output from the change I posted above (`ERR_print_errors_fp`).

Dantee296, 2019-09-16:
@nikias  okh i generated and compiled again using autoconf
and now its working with #ifdef HAVE_OPENSSL defined

ret = 0, length = 912
5328:error:0D07209B:asn1 encoding routines:ASN1_get_object:too long:../openssl-1.1.1b/crypto/asn1/asn1_lib.c:91:
5328:error:0D068066:asn1 encoding routines:asn1_check_tlen:bad object header:../openssl-1.1.1b/crypto/asn1/tasn_dec.c:1118:
5328:error:0D07803A:asn1 encoding routines:asn1_item_embed_d2i:nested asn1 error:../openssl-1.1.1b/crypto/asn1/tasn_dec.c:190:Type=ASN1_TIME
5328:error:0D08303A:asn1 encoding routines:asn1_template_noexp_d2i:nested asn1 error:../openssl-1.1.1b/crypto/asn1/tasn_dec.c:627:Field=notBefore, Type=X509_VAL
5328:error:0D08303A:asn1 encoding routines:asn1_template_noexp_d2i:nested asn1 error:../openssl-1.1.1b/crypto/asn1/tasn_dec.c:627:Field=validity, Type=X509_CINF
5328:error:0D08303A:asn1 encoding routines:asn1_template_noexp_d2i:nested asn1 error:../openssl-1.1.1b/crypto/asn1/tasn_dec.c:627:Field=cert_info, Type=X509
5328:error:0906700D:PEM routines:PEM_ASN1_read_bio:ASN1 lib:../openssl-1.1.1b/crypto/pem/pem_oth.c:33:
5328:error:140AB043:SSL routines:SSL_CTX_use_certificate:passed a null parameter:../openssl-1.1.1b/ssl/ssl_rsa.c:305:
ERROR: Could not connect to lockdownd, error code -5

Dantee296, 2019-09-16:
@nikias 

$ ideviceinfo.exe -d
06:46:18 lockdown.c:731 lockdownd_client_new(): device udid: 85335cb60ed6000491ff6e642a4ec6bd4b6abc19
06:46:18 lockdown.c:438 lockdownd_query_type(): called
06:46:18 property_list_service.c:132 internal_plist_send(): sending 284 bytes
06:46:18 service.c:144 service_send(): sending 4 bytes
06:46:18 service.c:144 service_send(): sending 284 bytes
06:46:18 property_list_service.c:137 internal_plist_send(): sent 284 bytes
06:46:18 property_list_service.c:138 internal_plist_send(): printing 284 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>ideviceinfo</string>
        <key>Request</key>
        <string>QueryType</string>
</dict>
</plist>
06:46:18 property_list_service.c:200 internal_plist_receive_timeout(): initial read=4
06:46:18 property_list_service.c:206 internal_plist_receive_timeout(): 297 bytes following
06:46:18 property_list_service.c:219 internal_plist_receive_timeout(): received 297 bytes
06:46:18 property_list_service.c:245 internal_plist_receive_timeout(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Request</key>
        <string>QueryType</string>
        <key>Type</key>
        <string>com.apple.mobile.lockdown</string>
</dict>
</plist>
06:46:18 lockdown.c:454 lockdownd_query_type(): success with type com.apple.mobile.lockdown
06:46:18 property_list_service.c:132 internal_plist_send(): sending 332 bytes
06:46:18 service.c:144 service_send(): sending 4 bytes
06:46:18 service.c:144 service_send(): sending 332 bytes
06:46:18 property_list_service.c:137 internal_plist_send(): sent 332 bytes
06:46:18 property_list_service.c:138 internal_plist_send(): printing 332 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>ideviceinfo</string>
        <key>Key</key>
        <string>ProductVersion</string>
        <key>Request</key>
        <string>GetValue</string>
</dict>
</plist>
06:46:18 property_list_service.c:200 internal_plist_receive_timeout(): initial read=4
06:46:18 property_list_service.c:206 internal_plist_receive_timeout(): 327 bytes following
06:46:18 property_list_service.c:219 internal_plist_receive_timeout(): received 327 bytes
06:46:18 property_list_service.c:245 internal_plist_receive_timeout(): printing 327 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Key</key>
        <string>ProductVersion</string>
        <key>Request</key>
        <string>GetValue</string>
        <key>Value</key>
        <string>10.3.3</string>
</dict>
</plist>
06:46:18 lockdown.c:509 lockdownd_get_value(): success
06:46:18 lockdown.c:520 lockdownd_get_value(): has a value
[... 49 lines cut ...]
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>Label</key>
        <string>ideviceinfo</string>
        <key>Request</key>
        <string>StopSession</string>
        <key>SessionID</key>
        <string>7C15371E-CB70-42A7-B160-4134E7665F2B</string>
</dict>
</plist>
06:46:18 idevice.c:438 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -108 (Unknown error)
06:46:18 service.c:167 service_receive_with_timeout(): could not read data
06:46:18 property_list_service.c:196 internal_plist_receive_timeout(): initial read failed!
06:46:18 lockdown.c:330 lockdownd_stop_session(): LOCKDOWN_E_PLIST_ERROR
14428:error:0D07209B:asn1 encoding routines:ASN1_get_object:too long:../openssl-1.1.1b/crypto/asn1/asn1_lib.c:91:
14428:error:0D068066:asn1 encoding routines:asn1_check_tlen:bad object header:../openssl-1.1.1b/crypto/asn1/tasn_dec.c:1118:
14428:error:0D07803A:asn1 encoding routines:asn1_item_embed_d2i:nested asn1 error:../openssl-1.1.1b/crypto/asn1/tasn_dec.c:190:Type=ASN1_TIME
14428:error:0D08303A:asn1 encoding routines:asn1_template_noexp_d2i:nested asn1 error:../openssl-1.1.1b/crypto/asn1/tasn_dec.c:627:Field=notBefore, Type=X509_VAL
14428:error:0D08303A:asn1 encoding routines:asn1_template_noexp_d2i:nested asn1 error:../openssl-1.1.1b/crypto/asn1/tasn_dec.c:627:Field=validity, Type=X509_CINF
14428:error:0D08303A:asn1 encoding routines:asn1_template_noexp_d2i:nested asn1 error:../openssl-1.1.1b/crypto/asn1/tasn_dec.c:627:Field=cert_info, Type=X509
14428:error:0906700D:PEM routines:PEM_ASN1_read_bio:ASN1 lib:../openssl-1.1.1b/crypto/pem/pem_oth.c:33:
14428:error:140AB043:SSL routines:SSL_CTX_use_certificate:passed a null parameter:../openssl-1.1.1b/ssl/ssl_rsa.c:305:
ERROR: Could not connect to lockdownd, error code -5

Dantee296, 2019-09-16:
@nikias  debug info
06:47:57 lockdown.c:731 lockdownd_client_new(): device udid: 85335cb60ed6000491ff6e642a4ec6bd4b6abc19
06:47:57 lockdown.c:438 lockdownd_query_type(): called
06:47:57 property_list_service.c:132 internal_plist_send(): sending 284 bytes
06:47:57 service.c:144 service_send(): sending 4 bytes
06:47:57 service.c:144 service_send(): sending 284 bytes
06:47:57 property_list_service.c:137 internal_plist_send(): sent 284 bytes
06:47:57 property_list_service.c:138 internal_plist_send(): printing 284 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>Request</key>
	<string>QueryType</string>
</dict>
</plist>
06:47:57 property_list_service.c:200 internal_plist_receive_timeout(): initial read=4
06:47:57 property_list_service.c:206 internal_plist_receive_timeout(): 297 bytes following
06:47:57 property_list_service.c:219 internal_plist_receive_timeout(): received 297 bytes
06:47:57 property_list_service.c:245 internal_plist_receive_timeout(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Request</key>
	<string>QueryType</string>
	<key>Type</key>
	<string>com.apple.mobile.lockdown</string>
</dict>
</plist>
06:47:57 lockdown.c:454 lockdownd_query_type(): success with type com.apple.mobile.lockdown
06:47:57 property_list_service.c:132 internal_plist_send(): sending 332 bytes
06:47:57 service.c:144 service_send(): sending 4 bytes
06:47:57 service.c:144 service_send(): sending 332 bytes
06:47:57 property_list_service.c:137 internal_plist_send(): sent 332 bytes
06:47:57 property_list_service.c:138 internal_plist_send(): printing 332 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>Key</key>
	<string>ProductVersion</string>
	<key>Request</key>
	<string>GetValue</string>
</dict>
</plist>
06:47:57 property_list_service.c:200 internal_plist_receive_timeout(): initial read=4
06:47:57 property_list_service.c:206 internal_plist_receive_timeout(): 327 bytes following
06:47:57 property_list_service.c:219 internal_plist_receive_timeout(): received 327 bytes
06:47:57 property_list_service.c:245 internal_plist_receive_timeout(): printing 327 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Key</key>
	<string>ProductVersion</string>
	<key>Request</key>
	<string>GetValue</string>
	<key>Value</key>
	<string>10.3.3</string>
</dict>
</plist>
06:47:57 lockdown.c:509 lockdownd_get_value(): success
06:47:57 lockdown.c:520 lockdownd_get_value(): has a value
06:47:57 userpref.c:195 userpref_read_system_buid(): using 307641261139469966264773636 as SystemBUID
06:47:57 property_list_service.c:132 internal_plist_send(): sending 430 bytes
[... 38 lines cut ...]
Nret = 0, length = 91206:47:57 idevice.c:827 idevice_connection_enable_ssl(): WARNING: Could not load RootCertificate
06:47:57 idevice.c:856 idevice_connection_enable_ssl(): ERROR in SSL_do_handshake: SSL_ERROR_ZERO_RETURN
06:47:57 lockdown.c:823 lockdownd_client_new_with_handshake(): Session opening failed.
06:47:57 lockdown.c:320 lockdownd_stop_session(): stopping session DE5182E8-9027-4BF9-A2A6-8B52FF4DCD86
06:47:57 property_list_service.c:132 internal_plist_send(): sending 363 bytes
06:47:57 service.c:144 service_send(): sending 4 bytes
06:47:57 service.c:144 service_send(): sending 363 bytes
06:47:57 property_list_service.c:137 internal_plist_send(): sent 363 bytes
06:47:57 property_list_service.c:138 internal_plist_send(): printing 363 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>Request</key>
	<string>StopSession</string>
	<key>SessionID</key>
	<string>DE5182E8-9027-4BF9-A2A6-8B52FF4DCD86</string>
</dict>
</plist>
06:47:57 idevice.c:438 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -108 (Unknown error)
06:47:57 service.c:167 service_receive_with_timeout(): could not read data
06:47:57 property_list_service.c:196 internal_plist_receive_timeout(): initial read failed!
06:47:57 lockdown.c:330 lockdownd_stop_session(): LOCKDOWN_E_PLIST_ERROR

Dantee296, 2019-09-16:
`06:47:57 lockdown.c:731 lockdownd_client_new(): device udid: 85335cb60ed6000491ff6e642a4ec6bd4b6abc19
06:47:57 lockdown.c:438 lockdownd_query_type(): called
06:47:57 property_list_service.c:132 internal_plist_send(): sending 284 bytes
06:47:57 service.c:144 service_send(): sending 4 bytes
06:47:57 service.c:144 service_send(): sending 284 bytes
06:47:57 property_list_service.c:137 internal_plist_send(): sent 284 bytes
06:47:57 property_list_service.c:138 internal_plist_send(): printing 284 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>Request</key>
	<string>QueryType</string>
</dict>
</plist>
06:47:57 property_list_service.c:200 internal_plist_receive_timeout(): initial read=4
06:47:57 property_list_service.c:206 internal_plist_receive_timeout(): 297 bytes following
06:47:57 property_list_service.c:219 internal_plist_receive_timeout(): received 297 bytes
06:47:57 property_list_service.c:245 internal_plist_receive_timeout(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Request</key>
	<string>QueryType</string>
	<key>Type</key>
	<string>com.apple.mobile.lockdown</string>
</dict>
</plist>
06:47:57 lockdown.c:454 lockdownd_query_type(): success with type com.apple.mobile.lockdown
06:47:57 property_list_service.c:132 internal_plist_send(): sending 332 bytes
06:47:57 service.c:144 service_send(): sending 4 bytes
06:47:57 service.c:144 service_send(): sending 332 bytes
06:47:57 property_list_service.c:137 internal_plist_send(): sent 332 bytes
06:47:57 property_list_service.c:138 internal_plist_send(): printing 332 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>Key</key>
	<string>ProductVersion</string>
	<key>Request</key>
	<string>GetValue</string>
</dict>
</plist>
06:47:57 property_list_service.c:200 internal_plist_receive_timeout(): initial read=4
06:47:57 property_list_service.c:206 internal_plist_receive_timeout(): 327 bytes following
06:47:57 property_list_service.c:219 internal_plist_receive_timeout(): received 327 bytes
06:47:57 property_list_service.c:245 internal_plist_receive_timeout(): printing 327 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Key</key>
	<string>ProductVersion</string>
	<key>Request</key>
	<string>GetValue</string>
	<key>Value</key>
	<string>10.3.3</string>
</dict>
</plist>
06:47:57 lockdown.c:509 lockdownd_get_value(): success
06:47:57 lockdown.c:520 lockdownd_get_value(): has a value
06:47:57 userpref.c:195 userpref_read_system_buid(): using 307641261139469966264773636 as SystemBUID
06:47:57 property_list_service.c:132 internal_plist_send(): sending 430 bytes
06:47:57 service.c:144 service_send(): sending 4 bytes
[... 38 lines cut ...]
06:47:57 idevice.c:856 idevice_connection_enable_ssl(): ERROR in SSL_do_handshake: SSL_ERROR_ZERO_RETURN
06:47:57 lockdown.c:823 lockdownd_client_new_with_handshake(): Session opening failed.
06:47:57 lockdown.c:320 lockdownd_stop_session(): stopping session DE5182E8-9027-4BF9-A2A6-8B52FF4DCD86
06:47:57 property_list_service.c:132 internal_plist_send(): sending 363 bytes
06:47:57 service.c:144 service_send(): sending 4 bytes
06:47:57 service.c:144 service_send(): sending 363 bytes
06:47:57 property_list_service.c:137 internal_plist_send(): sent 363 bytes
06:47:57 property_list_service.c:138 internal_plist_send(): printing 363 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>Request</key>
	<string>StopSession</string>
	<key>SessionID</key>
	<string>DE5182E8-9027-4BF9-A2A6-8B52FF4DCD86</string>
</dict>
</plist>
06:47:57 idevice.c:438 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -108 (Unknown error)
06:47:57 service.c:167 service_receive_with_timeout(): could not read data
06:47:57 property_list_service.c:196 internal_plist_receive_timeout(): initial read failed!
06:47:57 lockdown.c:330 lockdownd_stop_session(): LOCKDOWN_E_PLIST_ERROR
`

Dantee296, 2019-09-16:
```xml

06:47:57 lockdown.c:731 lockdownd_client_new(): device udid: 85335cb60ed6000491ff6e642a4ec6bd4b6abc19
06:47:57 lockdown.c:438 lockdownd_query_type(): called
06:47:57 property_list_service.c:132 internal_plist_send(): sending 284 bytes
06:47:57 service.c:144 service_send(): sending 4 bytes
06:47:57 service.c:144 service_send(): sending 284 bytes
06:47:57 property_list_service.c:137 internal_plist_send(): sent 284 bytes
06:47:57 property_list_service.c:138 internal_plist_send(): printing 284 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>Request</key>
	<string>QueryType</string>
</dict>
</plist>
06:47:57 property_list_service.c:200 internal_plist_receive_timeout(): initial read=4
06:47:57 property_list_service.c:206 internal_plist_receive_timeout(): 297 bytes following
06:47:57 property_list_service.c:219 internal_plist_receive_timeout(): received 297 bytes
06:47:57 property_list_service.c:245 internal_plist_receive_timeout(): printing 297 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Request</key>
	<string>QueryType</string>
	<key>Type</key>
	<string>com.apple.mobile.lockdown</string>
</dict>
</plist>
06:47:57 lockdown.c:454 lockdownd_query_type(): success with type com.apple.mobile.lockdown
06:47:57 property_list_service.c:132 internal_plist_send(): sending 332 bytes
06:47:57 service.c:144 service_send(): sending 4 bytes
06:47:57 service.c:144 service_send(): sending 332 bytes
06:47:57 property_list_service.c:137 internal_plist_send(): sent 332 bytes
06:47:57 property_list_service.c:138 internal_plist_send(): printing 332 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>Key</key>
	<string>ProductVersion</string>
	<key>Request</key>
	<string>GetValue</string>
</dict>
</plist>
06:47:57 property_list_service.c:200 internal_plist_receive_timeout(): initial read=4
06:47:57 property_list_service.c:206 internal_plist_receive_timeout(): 327 bytes following
06:47:57 property_list_service.c:219 internal_plist_receive_timeout(): received 327 bytes
06:47:57 property_list_service.c:245 internal_plist_receive_timeout(): printing 327 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Key</key>
	<string>ProductVersion</string>
	<key>Request</key>
	<string>GetValue</string>
	<key>Value</key>
	<string>10.3.3</string>
</dict>
</plist>
06:47:57 lockdown.c:509 lockdownd_get_value(): success
06:47:57 lockdown.c:520 lockdownd_get_value(): has a value
06:47:57 userpref.c:195 userpref_read_system_buid(): using 307641261139469966264773636 as SystemBUID
[... 41 lines cut ...]
06:47:57 lockdown.c:823 lockdownd_client_new_with_handshake(): Session opening failed.
06:47:57 lockdown.c:320 lockdownd_stop_session(): stopping session DE5182E8-9027-4BF9-A2A6-8B52FF4DCD86
06:47:57 property_list_service.c:132 internal_plist_send(): sending 363 bytes
06:47:57 service.c:144 service_send(): sending 4 bytes
06:47:57 service.c:144 service_send(): sending 363 bytes
06:47:57 property_list_service.c:137 internal_plist_send(): sent 363 bytes
06:47:57 property_list_service.c:138 internal_plist_send(): printing 363 bytes plist:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>ideviceinfo</string>
	<key>Request</key>
	<string>StopSession</string>
	<key>SessionID</key>
	<string>DE5182E8-9027-4BF9-A2A6-8B52FF4DCD86</string>
</dict>
</plist>
06:47:57 idevice.c:438 internal_connection_receive_timeout(): ERROR: usbmuxd_recv_timeout returned -108 (Unknown error)
06:47:57 service.c:167 service_receive_with_timeout(): could not read data
06:47:57 property_list_service.c:196 internal_plist_receive_timeout(): initial read failed!
06:47:57 lockdown.c:330 lockdownd_stop_session(): LOCKDOWN_E_PLIST_ERROR

```

nikias, 2019-09-16:
thanks. So I just ran the generated certificate through openssl on my mac:
```
$ openssl x509 -inform PEM -in rootcert.bin
unable to load certificate
4591302252:error:0DFFF09B:asn1 encoding routines:CRYPTO_internal:too long:/BuildRoot/Library/Caches/com.apple.xbs/Sources/libressl/libressl-22.200.4/libressl-2.6/crypto/asn1/asn1_lib.c:143:
4591302252:error:0DFFF066:asn1 encoding routines:CRYPTO_internal:bad object header:/BuildRoot/Library/Caches/com.apple.xbs/Sources/libressl/libressl-22.200.4/libressl-2.6/crypto/asn1/tasn_dec.c:1113:
4591302252:error:0DFFF03A:asn1 encoding routines:CRYPTO_internal:nested asn1 error:/BuildRoot/Library/Caches/com.apple.xbs/Sources/libressl/libressl-22.200.4/libressl-2.6/crypto/asn1/tasn_dec.c:208:Type=ASN1_TIME
4591302252:error:0DFFF03A:asn1 encoding routines:CRYPTO_internal:nested asn1 error:/BuildRoot/Library/Caches/com.apple.xbs/Sources/libressl/libressl-22.200.4/libressl-2.6/crypto/asn1/tasn_dec.c:627:Field=notBefore, Type=X509_VAL
4591302252:error:0DFFF03A:asn1 encoding routines:CRYPTO_internal:nested asn1 error:/BuildRoot/Library/Caches/com.apple.xbs/Sources/libressl/libressl-22.200.4/libressl-2.6/crypto/asn1/tasn_dec.c:627:Field=validity, Type=X509_CINF
4591302252:error:0DFFF03A:asn1 encoding routines:CRYPTO_internal:nested asn1 error:/BuildRoot/Library/Caches/com.apple.xbs/Sources/libressl/libressl-22.200.4/libressl-2.6/crypto/asn1/tasn_dec.c:627:Field=cert_info, Type=X509
4591302252:error:09FFF00D:PEM routines:CRYPTO_internal:ASN1 lib:/BuildRoot/Library/Caches/com.apple.xbs/Sources/libressl/libressl-22.200.4/libressl-2.6/crypto/pem/pem_oth.c:84:
```
The certificate is indeed malformed for some reason.

nikias, 2019-09-16:
Ok got it, your OpenSSL must be broken or missing a feature. Instead of `ASN1_UTCTIME` nodes it writes two invalid nodes there. It fails to write valid from and valid until fields. Check your OpenSSL configuration and rebuild.

Dantee296, 2019-09-16:
alright i ll try tomorrow , ll update my openSSL or try to debug 
ll let u know 

JFYI same exact code works totally fine on mac.
so it must be something with win32 env

Dantee296, 2019-09-17:
@nikias you were right about openssl 
msys base package installed openssl 1.1.1b 26feb 2019 which was causing problem
i reverted back to OpenSSL 1.0.2j  26 Sep 2016 . which fixed the issue for me

Dantee296, 2019-09-17:
@rcmpayne FYI

mexmer, 2019-09-17:
some time ago, i had this bug with openssl on windows, my fix was this macro
```
#define ASN1FIX(xx,yy) \
{\
	struct tm * _nowgmt;\
	__time64_t _gtime;\
	_time64(&_gtime);\
	_gtime += yy;\
	_nowgmt = _gmtime64(&_gtime);\
	char _timebuffer[30];\
	sprintf_s(_timebuffer, 30, "%02d%02d%02d%02d%02d%02dZ", _nowgmt->tm_year % 100, _nowgmt->tm_mon + 1, _nowgmt->tm_mday, _nowgmt->tm_hour, _nowgmt->tm_min, _nowgmt->tm_sec);\
	ASN1_TIME_set_string(xx, _timebuffer);\
}\
```
later in code i used it like this

```
		if (!ASN1_TIME_set(asn1time, time(NULL)))
			ASN1FIX(asn1time, 0);
```