libimobiledevice/libimobiledevice #804: error: Package requirements (libusbmuxd >= 1.1.0) were not met

yaseenmd, 2019-06-24:
While installing libimobiledevice on mac by running command "brew install libimobiledevice --HEAD"

Getting error as : checking for libusbmuxd >= 1.1.0... no
configure: error: Package requirements (libusbmuxd >= 1.1.0) were not met:

Requested 'libusbmuxd >= 1.1.0' but version of libusbmuxd is 1.0.10

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

[libimobiledevice log.txt](https://github.com/libimobiledevice/libimobiledevice/files/3319397/libimobiledevice.log.txt)

ddfreiling, 2019-07-04:
See https://github.com/flutter/flutter/issues/22595

Aquarian-Age, 2019-08-01:
How to solve this problem on Linux system?